Xinyuan-LilyGO/LilyGO-T-SIM7000G #206: Unable to get SDCard working. Bad Board?

tweathers-sec, 2022-12-28:
Unable able to get the SDCard working. Bad board? 

Using many variants of the sample code provided, I continue to get the following error. Yes, the SDCard is formatted to FAT.

`
[  1821][E][sd_diskio.cpp:802] sdcard_mount(): f_mount failed: (13) There is no valid FAT volume
SDCard MOUNT FAIL
`

Relevant Code Used - Snipped for brevity....

`
#include <SD.h>
#include <SPI.h>
#include <LittleFS.h>

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13

void setup()
{
    SPI.begin(SD_SCLK, SD_MISO, SD_MOSI, SD_CS);
    if (!SD.begin(SD_CS)) {
        Serial.println("SDCard MOUNT FAIL");
    } else {
        uint32_t cardSize = SD.cardSize() / (1024 * 1024);
        String str = "SDCard Size: " + String(cardSize) + "MB";
        Serial.println(str);
    }
}
`

Tig3rch3n, 2022-12-28:
Have you tried a different brand of SD Cards?
I had some Trouble with 'NoName' SDCards

They work under Windows/Linux/Android no Problem.
But for some reason they don't support "SPI Mode" so they are useless for any Arduino Projects.
Seytonic did a Video about the Topic a few days ago https://youtu.be/wNfpRwQgDG4

tweathers-sec, 2022-12-29:
Thanks for the recommendation, but I don't use off label cards for that specific reason. I have tested a number of different SanDisk and Kingston cards that work fine with my other ESP32's. For whatever, this board just doesn't like them.

Tig3rch3n, 2022-12-30:
Could you post the full Code?
I can test in Tonight/Tomorrow :)

tweathers-sec, 2022-12-30:
Sure thing... So far, I have just been using the Examples provided in this repo. 

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino

lewisxhe, 2023-01-03:
My test is normal.
1. Please try to add pinMode (SD_MISO, INPUT_PULLUP) before SPI.begin; Test whether it is normal
2. Make sure the SD card is FAT format and accessible on the computer
If it still doesn't work, I suspect that there is a problem with the SD card seat welding

[image]

Tig3rch3n, 2023-01-08:
> Sure thing... So far, I have just been using the examples provided in this repo.
> 
> https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino

Just tested it with my Version 20200415 Board, I had to Comment out the Modem Init (no IoT Sim yet)
But everything works out of the Box with a ext3 Formating 
I get 

> ========INIT========
>========SDCard Detect.======
>SDCard Size: 30436MB
>===========================

tweathers-sec, 2023-01-10:
Thanks all for the input. It was certainly a bad connection with the PCB. I ended up doing an exchange for a new board and my code is working without issue. Closing this out.

closed 2023-01-10 by tweathers-sec