Xinyuan-LilyGO/LilyGO-T-SIM7000G #207: Different connection speed for CAN communication ?

vkcorp, 2023-01-02:
As you may know I'm using SN65HVD230 with my TSIM7000G for CAN communication,
and I got some strange thing with it.

I have 2 target devices and I usually use 500kbps as below :

`CAN_cfg.speed = CAN_SPEED_500KBPS;`

it didn't make any problem if I used it with esp32-dev board, but in case of using TSIM7000G I have a problem !
One my device is ECUSIM2000 and it doesn't make any problem while using 500kbps+7000g and 500kbps+esp32.
https://www.obdsol.com/solutions/development-tools/obd-simulators/ecusim-2000/
The other one is real car and it has a problem with 500kbps even it's still working well with esp32-dev board.

After a few trials, I found I had to change the connection speed to 1000kbps.

`CAN_cfg.speed = CAN_SPEED_1000KBPS;`

But if I try 1000kbps with esp32, it's failed and no data from the car.

Is ther any reason I can expect with above situation ?
Why 7000G require to work with twice of connection speed, or ANY special cases ?

DevinCarpenter, 2023-02-04:
What?

Ok, let's keep in mind that the SIM7000G is just a cellular board that the ESP32 communicates with.

You are trying to communicate over a CAN protocol data bus from your car or the ECUSIM2000 OBD-2 simulator to the ESP32 on the Lilygo Board...correct?

The ESP32-Wrover-E is the ESP32 model we're working with, and it's definitely not the best one on the market, so it could be that the one you're using is just better.

What is the ESP32-dev board you are using. Do you have a model number?

Also - have you tried connecting to the esp32 over USB? The default baud rate for the ESP32 is 115200...so unless you change that - you're gonna get back garbage.

tanker5595, 2023-03-05:
Hi, have you solved this problem?

I am also having difficulty with the CAN  (TWAI) controller on LilyGO-T-SIM7000G.
I cannot get the board to communicate with a car or any other CAN BUS device. I have tried both SN65HVD230 and MCP2551 transceivers.

I've also tried multiple libraries but none of them worked so far.
eg. libraries
https://github.com/collin80/esp32_can
https://github.com/sandeepmistry/arduino-CAN

My intended purpose is to log some engine stats to an sd card and send the most critical togeather with GPS coordinates over GSM to a live dashboard.

vkcorp, 2023-03-07:
@DevinCarpenter 

> What is the ESP32-dev board you are using. Do you have a model number?
> 
> Also - have you tried connecting to the esp32 over USB? The default baud rate for the ESP32 is 115200...so unless you change that - you're gonna get back garbage.

I'm using below chip from Ali :

[1PCS ESP32 Development Board WiFi+Bluetooth Ultra-Low Power Consumption Dual Core ESP-32S ESP32-WROOM-32D ESP32-WROOM-32U ESP 32](https://www.aliexpress.com/item/1005001636295529.html)

Yes, usb connection speed is 115200, and CAN speed is 500KB.
Well, no problem in esp32 chips, but 7000G didn't.

@tanker5595 

> I cannot get the board to communicate with a car or any other CAN BUS device. I have tried both SN65HVD230 and MCP2551 transceivers.

In my cases, I don't have any problem on esp32 chips, but I had some troubles to do same things on 7000G.
No answer and no solution yet... But I recommend just try to change the CAN speed to various even it's not the value on the manual !

jichef, 2024-12-11:
> As you may know I'm using SN65HVD230 with my TSIM7000G for CAN communication, and I got some strange thing with it.
> 
> I have 2 target devices and I usually use 500kbps as below :
> 
> `CAN_cfg.speed = CAN_SPEED_500KBPS;`
> 
> it didn't make any problem if I used it with esp32-dev board, but in case of using TSIM7000G I have a problem !
> 
> FYI. One of my devices is ECUSIM2000 and it doesn't make any problem while using 500kbps+7000g and 500kbps+esp32. https://www.obdsol.com/solutions/development-tools/obd-simulators/ecusim-2000/ The other one is real car and it has a problem with 500kbps even it's still working well with esp32-dev board.
> 
> After a few trials, I found I had to change the connection speed to 1000kbps (only for 7000G).
> 
> `CAN_cfg.speed = CAN_SPEED_1000KBPS;`
> 
> But if I try 1000kbps with esp32, it's failed and no data from the car. esp32 only works with 500kbps as it did.
> 
> Is ther any reason I can expect with above situation ? Why 7000G require to work with twice of connection speed(compared to esp32), or do I have to consider ANY special cases with 7000G ?

Hello. Have you managed to communicate via CANbus?

I found by chance what is the origin of the communication problems via CANBus between different types of ESP32. Depending on the chip revision (3.1, 3.0, 2.0, 1.0...) the CANBus speed must be multiplied x2. That is, let's imagine that I have an ESP32 rev 1.0 that communicates the speed of the CAN vehicle port of 250KBPS; if I want to use a rev 3.x I must multiply the speed x2 for the communication to be effective (500KBPS/sec).

My problem now is with the lilygot7000g, which I am not able to communicate with other ESP32s. Could you share which GPIO pins you have used? You said that you were able to communicate with your ECUSIM2000. I am communicating at the same speed with other normal ESP32 (with the same chip revisions -both are rev 3.1) and I am not getting any information.

Thank you!

vkcorp, 2024-12-19:
Sorry but I didn't have any answer or hints from them about my strange situation, 
evenmore it's very old project and I don't have any code for now...

jichef, 2024-12-26:
Don't worry. I solved it a few days ago using the official TWAI library. CANbus speeds are no longer a problem. I'm now receiving CAN frames. Thanks for the reply anyway!