Xinyuan-LilyGO/LilyGO-T-SIM7000G #209: How can I receive and send SMS's?

Elianos11, 2023-01-12:
Can someone send me please code of sending and receiving SMS's?

vlszabolcs, 2023-01-18:
For sending: https://m2msupport.net/m2msupport/sms-at-commands/
It worked for me in Text mode. When sending, I was able to set an end-of-line character in the Termite application, which allowed me to send it.

droidblastnz, 2023-05-06:
//Example Read and look for "addsomething"
   if (response.indexOf("addsomething") != -1) {
      modem.sendAT("+CMGR=0");
      String messageContent = SerialAT.readString();  
      Serial.println("Message content: " + messageContent);  
      Serial.println("Report back");

//Check Balance
void Get_Balance()
{
  if (balance_requested) {  
 function
    //Serial.println("Balance has already been requested!");
    return;
  }
  
  String smsNumber = "xxx";                                    //Check Operator for code
  String smsMessage2 = "BAL";                                //Check Operator for details

  modem.sendAT("+CMGF=1");                               //Set the message format to text mode
  modem.waitResponse(1000L);
  modem.sendAT("+CNMI=2,1");                             //Enable SMS message notifications
  modem.waitResponse(1000L);

  String command = "+CMGS=\"" + smsNumber + "\"";
  modem.sendAT(command);
  modem.waitResponse(1000L);
  SerialAT.print(smsMessage2);                                //Write the SMS message
  modem.waitResponse(1000L);
  SerialAT.write(CR);                                                 //Send the Ctrl+Z character to terminate the message
  modem.waitResponse(5000L);
  pDBGln("+CMGS Sent BALANCE request");

  modem.sendAT("+CMGL=\"ALL\"");                     //List All messages   
  modem.waitResponse(5000L);                              //Wait up to 5 seconds for the message response and store it in the string variable
  modem.sendAT("+CMGL=\"REC UNREAD\"");    //Received unread messages e.g. new message
  modem.waitResponse(5000L);                             //Wait up to 5 seconds for the message response and store it in the string variable
  modem.sendAT("+CMGRD=0");                           //Send AT+CMGR command to read the SMS message at index 1 then deletes it
  data = SerialAT.readString();
  Serial.print("message: ");
  Serial.println(data);

  String balanceMessage = data.substring(data.indexOf("$"));    //Extract portion of message that contains balance
  balanceMessage.trim();                                                              //Remove any leading/trailing white space
  float balance = balanceMessage.substring(1).toFloat();            //Extract numeric value of balance
  Serial.print("Get balance: ");
  Serial.println(balance);
  credit = balance;
  //mqtt.publish(topicCredit, String(balance).c_str());
  balance_requested = true;                                   requested
}