Xinyuan-LilyGO/LilyGO-T-SIM7000G #210: Since the SIM7000 seems to support GPS and NMEA output over either USB or the UART3_TX/UART3_RX pins it might be a good idea to also connect (or at least breakout) those pins?

safesky-app, 2023-01-15:
Since the SIM7000 seems to support GPS and NMEA output over either USB or the UART3_TX/UART3_RX pins it might be a good idea to also connect (or at least breakout) those pins?

This way the standard RX/TX pins can be used as dedicated modem pins for AT commands while simultaneously using UART3 for NMEA data. If those pins aren’t connected it will be next to impossible to use the modem and GPS module at the same time. (NMEA can be output over USB but reading it out over USB using the esp32 seems to be unnecessarily complicating things)

Disclaimer: I haven’t actually used the SIM7000 module so perhaps somebody who actually has can comment whether there is another workaround.

_Originally posted by @ashley-schaeffer in https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/1#issuecomment-568289639_

safesky-app, 2023-01-16:
Unless I am mistaken, which PIN numbers should I configure in the code to connect to UART3?

vlszabolcs, 2023-01-18:
If I understand correctly, you want to use UART3 to read NMEA data. Unfortunately, as far as I know, the UART3 (50 and 49 pins) connections are not used. You can verify this by checking the schematic. If you also examine your board, you will see that there is no connection between the SIM7000 module's UART3 pinout and the ESP32. It may be possible to solve this issue by rewiring, if you have adequate soldering skills. If I remember correctly, the ESP32 still has a free UART port available or you can use a spare pin to implement software serial.

xyzzy42, 2023-04-14:
Keep in mind that SIM7000 is at 1.8V level, while all pins exposed on ESP32-WROVER module are 3.3 V.  So you will need a level shifter to connect UART3 to the ESP32.

ESP32 itself has a 1.8V power domain and some 1.8V pins, but they are all used internally in the WROVER module for flash and PSRAM and not exposed.

xyzzy42, 2023-04-20:
With a level converter I was able to wire pin 50 of the SIM70x0 module to pin 33 on the ESP32.  This lets me get NMEA data from the SIM70x0 module UART3 to the UART on the ESP32.  It works!

Lilygo could make this a lot easier by routing pin 50, the 1.8V rail, and an ESP32 pin to pads for a FET.  Populating the FET would create the necessary level converter to connect NMEA to ESP32.  Left unpopulated, it wouldn't use the ESP32 pin.

Doesn't have a fix yet, GPGGA will fill in when it does.

```
$GPGSV,3,1,10,06,55,124,45,11,27,173,36,12,39,294,31,14,07,108,30,1*6E
$GPGSV,3,2,10,17,34,058,28,19,63,055,41,24,54,247,22,32,07,326,20,1*6F
$GPGSV,3,3,10,03,09,045,00,25,07,293,00,1*66
$GLGSV,1,1,0,2*4B
$GAGSV,1,1,0,3*47
$QZGSV,1,1,0,5*4C
$BDGSV,1,1,0,4*40
$GPGSA,A,1,,,,,,,,,,,,,,,*1E
$GLGSA,A,1,,,,,,,,,,,,,,,*02
$GAGSA,A,1,,,,,,,,,,,,,,,*0F
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$QZGSA,A,1,,,,,,,,,,,,,,,*02
$GNGSA,A,1,,,,,,,,,,,,,,,*00
$GNVTG,,T,,M,,N,,K,N*32
$GNRMC,,V,,,,,,,,,,N,V*37
$GPGGA,,,,,,0,,,,,,,,*66
```