Xinyuan-LilyGO/LilyGO-T-SIM7000G #212: Lilygo 7000g gps not working

git-tiger, 2023-02-02:
I have the 7000g version, and tested the gpstest. Power to the antenna is working via the AT command.
Just that no lat/long data is shown.
Could it be the antenna is broken?

DevinCarpenter, 2023-02-04:
Possible but unlikely. I have quite a few of these units and every single one has worked great. What is the output you are getting?

Just so you know - This thing can take a while to connect. Being on the ground floor in a big building could prevent you from getting signal from the satellites....That caused problems for me. But being indoors in most places hasn't been an issue.

'+CGNSINF: 1,0,,,,,,,0,,,,,,4,,,,46,,' This is an example output of trying to get signal. I believe '46' is the number of GNSS satellites in range for location triangulation - though don't quote me.

This is along the lines of what you will get from the CGNSINF command when it connects.
'+CGNSINF: 1,1,20230126203118.000,38.897629,-77.036780,17.600,0.00,0.0,1,,1.0,1.3,0.9,,24,5,3,,43,,' 

Please be aware that the GPS actually has a cold start and a hot start as well. Cold start could take a while. For me it's about 20 seconds to 10 minutes - at which point I just reset the modem. Hot start could take less than 5 seconds. So connecting after you've already connected shortly before will always yield the fastest results.

git-tiger, 2023-02-06:
I got this to work now. Thanks to @DevinCarpenter 
Used AT command: AT+CGNSHOR=10 and now it is able to get the location.

closed 2023-02-06 by git-tiger

DevinCarpenter, 2023-02-08:
Glad to hear it. For future reference the comment detailing the commands I use [is here](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/202#issuecomment-1416930955)