Xinyuan-LilyGO/LilyGO-T-SIM7000G #214: Batch of devices with same IMEI in firmware

willclay9520, 2023-02-05:
I have 3 sim7000g modems that all show the exact same IMEI when calling for it in the firmware. The physical modem sticker is unique on each modem but firmware all show the same IMEI. I have not yet attempted to update firmware yet but will try that this week.

jjharlow, 2023-02-05:
I have the exact same issue with a recent batch of 12 I bought, at least 3 of the 6 I've checked have the same IMEI with a different physical modem sticker.

DevinCarpenter, 2023-02-08:
Hi guys.

Please use the command AT+SIMEI=<your imei>

I've tested this on my own SIM7000G and you can check the IMEI with the command AT+GSN or AT+CGSN to verify the change.

Also please check that your wireless provider receives the change of IMEI. If it's automatic updating it might take some time to propagate the change.

This command won't be found in the manual and comes from the SIM5320 chipset. Many commands are shared between modems, so it's best to keep that in mind. SIMCOM omits them because...they're unprofessional.

stefanidisgeorge, 2023-02-21:
is this solved? because i am about to order a big batch

jjharlow, 2023-02-21:
Thanks @DevinCarpenter that worked to updated the IMEI.

@stefanidisgeorge in terms of, can I update the IMEI, yes. Would I want to do this for a big batch of devices, probably not. Maybe someone else who purchased recently can chime in if they still see this.

DevinCarpenter, 2023-02-24:
@jjharlow Glad that worked.

I just bought a batch of 6. I can check and see if the IMEI is still a problem a little later today. 

Unfortunately though you're right that will be an issue for a lot of us having to take the time to update the IMEI with big batches...

@LilyGO Is it looking like the IMEI will have to be updated for these modules by hand indefinitely?

DevinCarpenter, 2023-02-27:
I just tested all 6 of my brand new units from AliExpress and the IMEIs were correct out of the box. Hopefully we won't end up seeing this issue again.

I chatted with Timothy Woo over at Botletics who develops his own SIM7000 boards and he said that SIMCOM sets the IMEI for every chip he buys - so this isn't something LilyGo would be the cause of. At least now we know we need to periodically check for this and hopefully it was just a fluke on SIMCOMs part.

DNS, 2023-12-29:
@DevinCarpenter is it possible to change the IMEI manually? 
I'm planning to buy LILYGO TTGO T-A7670E.

DevinCarpenter, 2023-12-30:
Yes, view AT+SIMEI comment above. Command may be different for the different device type but many commands are the same.

umerao, 2025-05-07:
> Hi guys.
> 
> Please use the command `AT+SIMEI=<your imei>`
> 
> I've tested this on my own SIM7000G and you can check the IMEI with the command `AT+GSN` or `AT+CGSN` to verify the change.
> 
> Also please check that your wireless provider receives the change of IMEI. If it's automatic updating it might take some time to propagate the change.
> 
> This command won't be found in the manual and comes from the SIM5320 chipset. Many commands are shared between modems, so it's best to keep that in mind. SIMCOM omits them because...they're unprofessional. But we already knew that hence the IMEIs.

I have purchased SIM900A from Aliexpress and your command `AT+SIMEI=<your imei>` gives error in my case. And `AT+GSN` command shows different IMEI that is printed on product tag.