Xinyuan-LilyGO/LilyGO-T-SIM7000G #219: VBAT + GND pins vs +/- battery pins on back of board

jjharlow, 2023-02-27:
I've read every issue regarding external batteries with these boards and this isn't clear to me. Is there a benefit to soldering my battery pack directly to the +/- pins on the back of the board (I removed the existing battery holder) vs using the VBAT & GND pins?

It sounds like there are some benefits, namely overcharge/discharge and retaining the use of the On/Off switch by connecting to the board via the battery pins on the back. If that is the case, absent any protection on my batteries and the VBAT not providing this, it seems like a big advantage. 
 
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/40#issuecomment-661050487

However, another post suggest that the behavior of the on/off switch and associated logic may be questionable and bypassing it was a better choice.

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/65#issuecomment-770247748

So my question, is there any actual recommendation or is it situational?

Thanks.

jjharlow, 2023-03-08:
Well, for me it's going to come down to leveraging the onboard solar charger, which means I will be soldering to the back of the board.

closed 2023-03-08 by jjharlow

fgnievinski, 2023-06-06:
related information:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/37#issuecomment-674332698