Xinyuan-LilyGO/LilyGO-T-SIM7000G #223: Read battery?

killerbeat12, 2023-03-05:
How could i read battery voltage from the battery that could be connected on the board?

jjharlow, 2023-03-17:
That's covered pretty well here: 

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/104#issuecomment-833887214

jjharlow, 2023-03-23:
You have to add that function, her covered it here.

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/issues/82#issuecomment-1198896521

procopio, 2023-07-25:
Sorry to came back to this topic, but I'm getting this strange reads from your function, what could be wrong please?

I'm feeding fixed 4v and 5A in battery socket with a power source.

******* readVcc voltageBuffer = 2460.00, 2463.00, 2465.00, 2466.00, 2466.00, 2471.00, 2480.00, 2480.00, 2481.00, 2481.00, 2484.00, 2484.00, 2486.00, 2492.00, 2496.00, !
******* readVcc voltageBuffer = 2453.00, 2455.00, 2458.00, 2462.00, 2467.00, 2467.00, 2467.00, 2468.00, 2473.00, 2473.00, 2483.00, 2487.00, 2490.00, 2491.00, 2497.00, !
******* readVcc voltageBuffer = 2461.00, 2465.00, 2466.00, 2469.00, 2470.00, 2474.00, 2477.00, 2478.00, 2481.00, 2482.00, 2483.00, 2485.00, 2487.00, 2496.00, 2496.00, !
******* readVcc voltageBuffer = 2461.00, 2461.00, 2463.00, 2465.00, 2467.00, 2471.00, 2474.00, 2479.00, 2480.00, 2481.00, 2482.00, 2485.00, 2486.00, 2487.00, 2496.00, !
******* readVcc voltageBuffer = 2466.00, 2474.00, 2475.00, 2479.00, 2480.00, 2482.00, 2482.00, 2485.00, 2486.00, 2488.00, 2489.00, 2494.00, 2494.00, 2495.00, 2495.00, !
******* readVcc voltageBuffer = 2457.00, 2465.00, 2475.00, 2475.00, 2475.00, 2478.00, 2480.00, 2480.00, 2480.00, 2480.00, 2481.00, 2481.00, 2481.00, 2481.00, 2495.00, !
******* readVcc voltageBuffer = 2461.00, 2465.00, 2468.00, 2470.00, 2471.00, 2474.00, 2475.00, 2480.00, 2480.00, 2484.00, 2485.00, 2488.00, 2490.00, 2493.00, 2493.00, !
******* readVcc voltageBuffer = 2459.00, 2465.00, 2467.00, 2467.00, 2469.00, 2475.00, 2475.00, 2477.00, 2478.00, 2480.00, 2482.00, 2487.00, 2493.00, 2494.00, 2496.00, !
******* readVcc voltageBuffer = 2457.00, 2462.00, 2467.00, 2467.00, 2470.00, 2471.00, 2474.00, 2476.00, 2477.00, 2481.00, 2489.00, 2492.00, 2493.00, 2495.00, 2498.00, !
******* readVcc voltageBuffer = 2455.00, 2459.00, 2459.00, 2461.00, 2469.00, 2470.00, 2471.00, 2474.00, 2478.00, 2480.00, 2480.00, 2490.00, 2491.00, 2498.00, 2499.00, !
******* readVcc voltageBuffer = 2455.00, 2462.00, 2464.00, 2471.00, 2471.00, 2471.00, 2475.00, 2476.00, 2479.00, 2479.00, 2482.00, 2486.00, 2486.00, 2489.00, 2491.00, !
******* readVcc voltageBuffer = 2462.00, 2465.00, 2469.00, 2469.00, 2469.00, 2469.00, 2475.00, 2476.00, 2479.00, 2479.00, 2480.00, 2481.00, 2483.00, 2492.00, 2494.00, !
******* readVcc voltageBuffer = 2465.00, 2465.00, 2466.00, 2469.00, 2469.00, 2470.00, 2474.00, 2475.00, 2475.00, 2475.00, 2479.00, 2480.00, 2483.00, 2485.00, 2491.00, !
******* readVcc voltageBuffer = 2464.00, 2467.00, 2469.00, 2469.00, 2471.00, 2471.00, 2474.00, 2476.00, 2478.00, 2479.00, 2479.00, 2480.00, 2481.00, 2481.00, 2488.00, !
******* readVcc voltageBuffer = 2466.00, 2469.00, 2471.00, 2474.00, 2477.00, 2480.00, 2480.00, 2481.00, 2483.00, 2486.00, 2486.00, 2487.00, 2495.00, 2496.00, 2496.00, !
******* readVcc Read_buffer, bat, BP = 37161 , 4.35 , 109.06
batteryVoltage = 109%

```
void sortData(float data[], int8_t size) {             // Sorts all data small to large
    float swapper;
    for (int8_t i = 0; i < (size - 1); i++) {
        for (int8_t o = 0; o < (size - (i + 1)); o++) {
            if (data[o] > data[o + 1]) {
                swapper = data[o];
                data[o] = data[o + 1];
                data[o + 1] = swapper;
            }
        }
    }
}

float mapBatt(float x, float in_min, float in_max, float out_min, float out_max) {
    float ret = (x - in_min) * (out_max - out_min) / (in_max - in_min) + out_min;
    if (ret < 0) { ret = 0; }
    return ret;
}

int readVcc() {         // reads and returns the battery voltage
    int VReads = 15;
    float voltageBuffer[VReads];
    uint32_t Read_buffer = 0;
    for (int x = 0; x < VReads; x++) {
        for (int i = 0; i < VReads; i++) {
            voltageBuffer[i] = (uint32_t)analogRead(BATTERY_POWER_PIN);
        }
        sortData(voltageBuffer, VReads);

        DEBUG_print(F("******* readVcc voltageBuffer = "));
        for (size_t i = 0; i < VReads; i++) {
            DEBUG_print(voltageBuffer[i]);
            DEBUG_print(F(", "));
        }
        DEBUG_println(F("!"));

        Read_buffer += (voltageBuffer[(VReads - 1) / 2]);        
    }

    float bat = (((float)(Read_buffer / VReads) / 4096) * 3600 * 2) / 1000;
    float BP = mapBatt(bat, 2.5, 4.2, 0, 100);     // get battery voltage as a percentage 0-100%

    DEBUG_print(F("******* readVcc Read_buffer, bat, BP = "));
    DEBUG_print((long)Read_buffer);
    DEBUG_print(F(" , "));
    DEBUG_print(bat);
    DEBUG_print(F(" , "));
    DEBUG_println(BP);

    return (int)BP;
}
```

fgnievinski, 2024-01-15:
a better answer:
https://github.com/Xinyuan-LilyGO/TTGO-T-Display/issues/35#issuecomment-749511521