Xinyuan-LilyGO/LilyGO-T-SIM7000G #225: Azure IOTHub

Gavo4, 2023-04-03:
Hi There,

I am trying to connect to the Azure IOT using the example code. Looks like there is a TCIP issue with the existing code. The sample code also look a little old. Is there a new update?

Regards,

Gavin

Gavo4, 2023-04-07:
I have re-followed the example code for the Azure IOT Hub and I keep getting stuck here, any ideas?

Modem Connected to Rogers' LTE (channel--> 38) CAT-M (preferred network). TLE CAT-M OK
 > Modem On --> True
Wake up modem from sleep
Modem hardware reset
 > IoT Hub
Info: Initializing SNTP

MIKHANYA, 2023-07-26:
@ejri Hello, can you please post your code for ESP32 + SIM7000G for IoT Hub to which you came two years ago, is it still working, is it relevant?
https://github.com/Azure/azure-iot-sdk-c/issues/1914#issuecomment-808452681

ejri, 2023-07-26:
@MIKHANYA yes. i posted it in examples:

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino

MIKHANYA, 2023-07-28:
@Gavo4 I also had problems with my LTE connection, I solved it by setting commands from another example. https://github.com/botletics/Botletics-SIM7000/blob/main/examples/ESP32_LTE_Demo/ESP32_LTE_Demo.ino
 (Maybe this is a problem with my SIM card support of this communication standard)
@ejri Thanks for the code example, IOT HUB connection part works grate. 
But there is a nuance, You need to use an older version of the ESP32 kernel. More about this issue:
https://www.reddit.com/r/esp32/comments/112u7co/esp32_azureiot_arduino_doesnt_work_on_newer_esp32/
https://github.com/VSChina/ESP32_AzureIoT_Arduino/issues/24#issuecomment-1087517787

Gavo4, 2023-07-28:
@MIKHANYA I will try this over the weekend. I tried a number of samples and could get nothing to work. No issue to set it up via Wi-Fi but with LTE I could not get it to work at all. I will check this out and advise

MIKHANYA, 2023-08-04:
> I have re-followed the example code for the Azure IOT Hub and I keep getting stuck here, any ideas?
> 
> Modem Connected to Rogers' LTE (channel--> 38) CAT-M (preferred network). TLE CAT-M OK
> 
> > Modem On --> True
> > Wake up modem from sleep
> > Modem hardware reset
> > IoT Hub
> > Info: Initializing SNTP

I changed esp32 board, to newer form China (it has "-32E" ending code). And I get STUCKED in that place too.

ejri, 2023-08-04:
Initializing SNTP error means that you're not really connected to a network. 

the way the esp32 hardware architecture is set up is that you must enable include the wifi.h library and enable the wifi module to power up the rest of the esp module. Otherwise, it will not connect to any network, including cellular. Insane architecture because it defeats the purpose of power management. 

In other words, 
1. turn on board
2. turn on wifi
3. do something (in your case: turn on cellular modem, and connect to cellular). 

I have not tested this on new versions of the board (if there are any).

Gavo4, 2023-08-06:
@ejri  I am still having connection issues to the IOT Hub

All the AT Commands are working for the exception of the Error 

Error: Time:Sun Aug  6 14:13:46 2023 File:.pio/libdeps/esp32doit-devkit-v1/ESP32 Azure IoT Arduino/src/az_iot/iothub_client/src/iothub_client_ll.c Func:IoTHubClient_LL_CreateFromConnectionString Line:824 deviceId is not found

Any idea what this can be?, below is the terminal output

AT
OK
ATE0
ATE0
OK
AT+CMEE=0

OK
AT+CLTS=1

OK
AT+CBATCHK=1

OK
AT+CPIN?

+CPIN: READY

OK
Step 2: [OK] was able to open modem
ATI

SIM7000G R1529

OK
Step 3: Modem details:
SIM7000G R1529
Waiting for network...
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,2

OK
AT+CGREG?

+CGREG: 0,2

OK
AT+CEREG?

+CEREG: 0,1

OK
 success
AT+CEREG?

+CEREG: 0,1

OK
Network connected
Step 4: Waiting for network...AT+CEREG?

+CEREG: 0,1

OK
Found network: [OK]
Step 5: About to set network mode to LTE Only 38: AT+CNMP=38

OK
Step 6: About to set network mode: to CAT=MAT+CMNB=3

OK
Waiting for network...AT+CEREG?

+CEREG: 0,1

OK
 OK
Signal quality:AT+CSQ

+CSQ: 23,99

OK
23
Step 7: Connecting to Rogers APN at LTE Mode Only (channel--> 38): 
lte.vodacom.za
AT+CIPSHUT

SHUT OK
AT+CGATT=0

OK
AT+SAPBR=3,1,"Contype","GPRS"

OK
AT+SAPBR=3,1,"APN","lte.vodacom.za"

OK
AT+CGDCONT=1,"IP","lte.vodacom.za"

OK
AT+CGATT=1

OK
AT+CGACT=1,1

OK
AT+SAPBR=1,1

OK
AT+SAPBR=2,1

+SAPBR: 1,1,"100.108.194.200"

OK
AT+CIPMUX=1

OK
AT+CIPQSEND=1

OK
AT+CIPRXGET=1

OK
AT+CSTT="lte.vodacom.za","",""

OK
AT+CIICR

OK
AT+CIFSR;E0

100.108.194.200

OK
AT+CGATT?

+CGATT: 1

OK
AT+CIFSR;E0

100.108.194.200

OK
Step 8: Connected to network: [OK]
AT+CIFSR;E0

100.108.194.200

OK
Step 9: Local IP: 100.108.194.200
AT+CGNSPWR=1

OK
AT+GSN

869951035890064

OK
Step 10: IMEI: 869951035890064
Modem Connected to Rogers' LTE (channel--> 38) CAT-M (preferred network). TLE CAT-M OK
 > Modem On --> True
 > IoT Hub
Info: Initializing SNTP
Info: SNTP initialization complete
Info: IoT Hub SDK for C, version 1.1.23
Error: Time:Sun Aug  6 14:13:46 2023 File:.pio/libdeps/esp32doit-devkit-v1/ESP32 Azure IoT Arduino/src/az_iot/iothub_client/src/iothub_client_ll.c Func:IoTHubClient_LL_CreateFromConnectionString Line:824 deviceId is not found
Initializing IoT hub failed.
modemConnected --> True51635
reading MPU sensor data
[ 51636][E][Wire.cpp:422] beginTransmission(): could not acquire lock
[ 51636][E][Wire.cpp:526] write(): NULL TX buffer pointer
[ 51640][E][Wire.cpp:448] endTransmission(): NULL TX buffer pointer
[ 51646][E][Wire.cpp:481] requestFrom(): NULL buffer pointer
[ 51652][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51657][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51662][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51667][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51672][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51677][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51682][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51687][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51692][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51697][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51702][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51707][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51712][E][Wire.cpp:557] read(): NULL RX buffer pointer
[ 51717][E][Wire.cpp:557] read(): NULL RX buffer pointer
0/0/0
{"DeviceID": "roll1":1, "pitch1":0, "yaw1":0}
Going to sleep now with modem turned off
AT+CSCLK=0

OK
AT+CPOWD=1

Gavo4, 2023-08-06:
@ejri  in this part below  have just used the connection string from the Shared access policies on the IOT Hub under the IOTHubOwner  primary connection string.  I assume we just need to enter that?

/// Primaty Connection String///
/*String containing Hostname, Device Id & Device Key in the format:                         */
/*  "HostName=<host_name>;DeviceId=<device_id>;SharedAccessKey=<device_key>"                */
/*  "HostName=<host_name>;DeviceId=<device_id>;SharedAccessSignature=<device_sas_token>"    */
static const char *connectionString =

ejri, 2023-08-06:
it seems to connect to the internet just fine. Try making an http request to anything other than Azure to be certain. 
e.g. fetch weather data from openweatherapi or something like that. 

It seems that the issue is at the IoT Hub. Double check the connection string, try connecting in different ways. for example using the host name approach. 

On the Azure end, make sure that data can be uploaded properly. For example, it will only accept data in very specific timedate formats. Make sure that the data type you upload is consistent. For example, if you are uploading x = float, dont change it to x = string kind of thing, and see if that makes a difference. 

You can also try to connect to IoT Hub from somewhere else, for example python api to make sure that data is there are no issues uploading data to iot hub.

MIKHANYA, 2023-08-07:
> Initializing SNTP error means that you're not really connected to a network.
> 
> the way the esp32 hardware architecture is set up is that you must enable and include the wifi.h library and enable the wifi module to power up the rest of the esp module. Otherwise, it will not connect to any network, including cellular. Insane architecture because it defeats the purpose of power management.
> 
> In other words,
> 
> 1. turn on board
> 2. turn on wifi
> 3. do something (in your case: turn on cellular modem, and connect to cellular).
> 
> I have not tested this on new versions of the board (if there are any).

@ejri Sorry, of course, this may seem rude, but it seems to me that your tests were not clean.
Last time I wrote to you that the code works, but in fact it turned out that it worked all the time through Wi-Fi.
Let me prove it to you.
Before I used the code through wi-fi, examples from Azure. Later, I uploaded code that uses cellular [Arduino_Azure_IoTHub](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino). And it worked thanks to the change of version to 1.0.6 In the board manager for esp 32 in ARDUINO ide. I wanted to flash another new esp32 device that did not use the wifi code, and it hung at the moment of: `Info: Initializing SNTP`. And then I began to suspect something, I decided to clear the flash using(https://espressif.github.io/esptool-js/) on the first working device **LilyGO-T-SIM7000G** and upload code [Arduino_Azure_IoTHub](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino) with my connection string. And as a result, the device also hung in the same place`Info: Initializing SNTP`. 
Based on this, I conclude that this code has never transmitted data over a cellular connection. And you, we've all been misled.

ejri, 2023-08-07:
I mean, I have been using those in the field where there is no wifi, as it is running only on cellular. I have been using those for over 3years now.

Check what others have posted. It literally shows that the module connects to the internet. 

As I said in an earlier comment, how you set up IoT Hub and the type of data you upload matters. Also check if you can connect to any other website.

MIKHANYA, 2023-08-08:
> I mean, I have been using those in the field where there is no wifi, as it is running only on cellular. I have been using those for over 3years now.
>
> Check what others have posted. It literally [...]

My code is no different from yours, I just commented out everything related to reading sensors. And commented this lines: 
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/196c3aef13fe7b31b5a00fa97dbd9bb959f56bd5/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino#L284
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/196c3aef13fe7b31b5a00fa97dbd9bb959f56bd5/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino#L290
all I added is my connection string. Please tell me what is your test setup, and what versions of libraries did you use ?

MIKHANYA, 2023-08-11:
@lewisxhe @LilyGO Can you guys help? Check if this code is working now and what versions of the libraries should be used?