Xinyuan-LilyGO/LilyGO-T-SIM7000G #227: Board stop transmiting if battery is under 4 volts

Gabriel-Gardin, 2023-04-10:
Hello. I am having problems when trying to use the board with not totally full batteries.

The schematic says that the under voltage protection voltage is defined from the DW01 chip(2.4 Volts) 

I am testing with a battery which is at 3.86 volts, everything starts to work but then it doesn't finish transmitting and the boards is turned off until I press the pwr button, which, as per my understanding, resets the DW01 chip and the board comes back on.

Is this the expected behaviour? Shouldn't it be working with almost 3.9 volts?

Gabriel-Gardin, 2023-04-10:
Ok, so I have debugged a little more an it seems that the problem is that when the board starts transmitting some heavy ripple starts on the battery, which cannot stand the high current surges. 

In the image bellow you can se the ripple that is causing the problems... 

[image]

Gabriel-Gardin, 2023-04-14:
Ok... so I have figured out what the problem is.

IN gprs class 12, when transmitting data the modem can consume bursts of 2.3msecs and 4.616ms period when the current can go as up to 2A. This is way more than a common 18650 battery can give, therefore it's a common practice to add supercapacitors to help the battery with this high current bursts.

So I added a 5.5V 1F super cap in parallel with the battery and now, everything is working(currently the battery is at 3.69V) working perfectly...

More info about powering GPRS modems with battery here:

https://www.tecategroup.com/white_papers/ppr_2003_GPRS-perf.pdf

https://www.ti.com/lit/ug/tiducg0/tiducg0.pdf?ts=1681163344041&ref_url=https%253A%252F%252Fwww.google.com%252F#:~:text=The%20typical%20voltage%20of%20a,about%2020%20mA%2F1000%20mAh

https://usermanual.wiki/Pdf/QuectelLowPowerDesignUserGuideV13.1941735194

Gabriel-Gardin, 2023-04-14:
So to close this I would suggest adding a super cap in the board... 5.5V x 1F seems to be good enough for Li-Ion 18650 batteries.

closed 2023-04-14 by Gabriel-Gardin

fgnievinski, 2023-07-07:
@Gabriel-Gardin could you post a photo, please? I'm trying to see how close to the board the capacitor needs to be. Also, did you put a resistor or a switch to avoid short circuiting when the capacitor is empty? We've tried a full faraday supercapacitor and it helps when the battery is fully charged, but not below 4V. Many thanks for your help.

valdobb, 2024-01-26:
I rather met using  Super pulse battery capacitor (SPC)  and Hybrid Layer Capacitors (HLC) rather than the usual capacitors.