Xinyuan-LilyGO/LilyGO-T-SIM7000G #228: Schematic for T-SIM7070G

xyzzy42, 2023-04-14:
There does not seem to be a schematic for this board.

It looks like it is similar to T-SIM7000G.   But are there differences?  For instance, does it have GPS antenna power on GPIO4 of SIMCOM 7070G module?

xyzzy42, 2023-04-14:
Tracing the connections on the board and tests show that GPS antenna power is on GPIO **5** on the TSIM-7070G.

lewisxhe, 2023-04-21:
yes i found that too it's different from 7000G, it's been fixed here, thanks
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/commit/3ea23765b752bb513105a3ea6e81696418cb60a1

closed 2023-04-21 by lewisxhe

xyzzy42, 2023-04-21:
It might be better to use `AT+CGPIO=0,48,1,1` instead of the `SGPIO` command.  The **C**GPIO version uses the pin number instead of the gpio number.  The pin is 48 on both SIM7000 and SIM7070 so the same command will work on both.  This is simpler and also does not unnecessarily control the other GPIO not used for GPS power.

There are also a bunch of places not fixed for SIM7070.

Like the main documentation, https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/e04c7a11ddd73bf550121aeabd3b11e81896c4dc/README.MD?plain=1#L42

Also here:

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/e04c7a11ddd73bf550121aeabd3b11e81896c4dc/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino#L66

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/e04c7a11ddd73bf550121aeabd3b11e81896c4dc/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino#L126

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/e04c7a11ddd73bf550121aeabd3b11e81896c4dc/examples/MicroPython_LoBo/gps.py#L67

And I think firmware `T-sim7000X-AllFunctions.bin` does not have the command.

lewisxhe, 2023-04-22:
Thanks for the reminder, it has been fixed.