Xinyuan-LilyGO/LilyGO-T-SIM7000G #229: Read SIM7000G Power status from ESP32

markjandrews, 2023-04-19:
Currently SIM7000G pin 66 (STATUS) only drives an LED. Is there any potential to get this pin able to drive an input on the ESP32 so we can test whether the the SIM7000G has power. At the moment there is no way to know way to actually know whether the "toggle" power button is turning the SIM7000G on or off and it can get out of sync.

DevinCarpenter, 2023-04-24:
Getting the power status of the SIM7000G is frustrating for me too. I use the AT command to see if the modem responds. If it responds - then I know it has power, otherwise I toggle the power.

That solution has worked reliably for me.

shanedoolane, 2023-10-03:
@DevinCarpenter @markjandrews did you good way to determine if power is on but modem is in a frozen/unresponsive state? If the modem doesn't respond to AT, it could be:

1. Stuck
2. Off

If in state 1, we want to power cycle it. State 2 maybe we try the same? 
```cpp
// Turn off
pinMode(MODEM_PWKEY, OUTPUT);
digitalWrite(MODEM_PWKEY, LOW);
delay(1500); // Datasheet T_on mintues = 1.25S Lilygo
digitalWrite(MODEM_PWKEY, HIGH);
// Turn back on
pinMode(MODEM_PWKEY, OUTPUT);
digitalWrite(MODEM_PWKEY, HIGH); // https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/196c3aef13fe7b31b5a00fa97dbd9bb959f56bd5/examples/Arduino_TinyGSM/AllFunctions/AllFunctions.ino#L74C1-L74C1
delay(1000);                     
digitalWrite(MODEM_PWKEY, LOW);

// check AT
modem.testAT

// graceful shutdown
modem.sendAT("+CPOWD=1")
```

DevinCarpenter, 2023-12-30:
@shanedoolane as far as finding a way that works - I toggle the power if there is no response. This has worked well for me, generally sending out a number of AT commands works to test for responsiveness.