Xinyuan-LilyGO/LilyGO-T-SIM7000G #230: SIM7000E stops working after an hour

nkkn87, 2023-04-20:
I am using SIM7000E with telstra network here in australia and using http get. I am able to connect network, send data to server, however, after an hour, I encountered that SIM stops working. (i.e, only red power light is on and the network status light is totally off), I can't even send AT commands, there will be no response from SIM. I will to power cycle the SIM module, then it starts working again. Anybody has encountered this before and any idea to solve this?

Thanks in advance.

lewisxhe, 2023-04-21:
This situation should be caused by unstable power supply, please check whether the power supply is normal

nkkn87, 2023-05-04:
I am using bench top power supply and it is stable I guess. it stops responding after sending at+httpaction=0. it can be after 45 mins or an hour.

HookedonIOT, 2023-05-07:
I have consistent running in 1 hour deep sleep here in Australia on Telstra nbiot using m2m one sim. Using 18650 battery on the board. Latest version board. Sending data to Blynk. I experienced the no connect red blink last night so investigating if battery was too low. Don’t have multimeter with me but will check on next red event. Wondering as im not using gps for my project whether I need to code the pin 5 high to save power drain though?

DevinCarpenter, 2023-08-01:
Have you tried using a different M2M provider? I had an issue with Sierra Wireless disconnecting the modem from the tower repeatedly. After I switched the provider I haven't had a single problem.