Xinyuan-LilyGO/LilyGO-T-SIM7000G #233: MQTT example seems to fall over if using WIFI connection

bucklevision, 2023-05-05:
Compilation error: 'TinyGsm' {aka 'class TinyGsmSim7000'} has no member named 'networkConnect'; did you mean 'isNetworkConnected'?

droidblastnz, 2023-05-06:
To fix this error, you need to change the function call from networkConnect to isNetworkConnected. Alternatively, you could add a member function called networkConnect to the TinyGsm class that performs the desired network connection functionality.

Here's an example of how you could modify your code to use the isNetworkConnected function

```
#include <TinyGsm.h>

TinyGsm modem;

void setup() {
  // initialize the modem here
  modem.init();

  // check if the network is already connected
  if (modem.isNetworkConnected()) {
    Serial.println("Network is already connected");
  } else {
    Serial.println("Connecting to network...");
    if (modem.networkConnect()) {
      Serial.println("Network connected");
    } else {
      Serial.println("Network connection failed");
    }
  }
}

void loop() {
  // your code here
}
```

JohnnyLiu2886, 2024-02-28:
> To fix this error, you need to change the function call from networkConnect to isNetworkConnected. Alternatively, you could add a member function called networkConnect to the TinyGsm class that [...]

"networkConnect" and "isNetworkConnected" are not the same function. You cannot replace "networkConnect" using "isNetworkConnected".

"networkConnect" is a function to connect the lilygo to WIFI.

However, I have no idea why "TINY_GSM_MODEM_SIM7600" and "TINY_GSM_MODEM_SIM7000" are not supporting "networkConnect".

The problem is that no body mentions whether can the LilyGo connect to wifi instead of using sim card network.