Xinyuan-LilyGO/LilyGO-T-SIM7000G #234: NTP Fails error 61

Michaelcombs, 2023-05-08:
NTP always fails with error 61.

 I'm connecting via LTE and CAT-M. I am able to open TCP connections for MQTT and streaming. I have included my AT command sequence. I am using TinyGSM with TINY_GSM_MODEM_SIM7000SSL. 

AT+CFUN=1
OK
AT+CNMP=38
OK
AT+CMNB=1
OK
AT+CEREG?
+CEREG: 0,0
OK
AT+CGREG?
+CGREG: 0,1
OK
AT+CEREG?
+CEREG: 0,0
OK
AT+CGREG?
+CGREG: 0,1
OK
AT+CNACT=0
ERROR
AT+CGDCONT=1,"IP","IoT.simon"
OK
AT+CGATT=1
OK
AT+CNCFG=1,"IoT.simon"
OK
AT+CNACT=1,"IoT.simon"
OK
+APP PDP: DEACTIVE
AT+CNACT=1,"IoT.simon"
OK
+APP PDP: ACTIVE
DST: 1
*PSUTTZ: 23/05/08,03:16:23","-20",1
AT+CNTPCID=1
OK
AT+CNTP="pool.ntp.org",0
OK
AT+CNTP
OK
+CNTP: 61

trentp-igor, 2023-05-14:
It doesn't work for me either.

I tried sending UDP packets to the NTP port of a server I can monitor, with `at+caopen=9,0,"UDP","server",123,1` and `at+casend=9,length`, and this worked.  So sending NTP packets can apparently work.

But when I use `at+cntp` I get nothing on the server.

trentp-igor, 2023-05-15:
Got it to work.  I'm using the SIM7070G model.  Maybe it's necessary to manually specify the pdp context in the cntp command?

```
AT+CNACT=1,1
OK
+APP PDP: 1,ACTIVE

AT+CNACT?
+CNACT: 0,0,"0.0.0.0"
+CNACT: 1,1,"100.198.223.226"
+CNACT: 2,0,"0.0.0.0"
+CNACT: 3,0,"0.0.0.0"

AT+CNTP="time.cloudflare.com",-28,1,2
OK

AT+CNTP
OK
+CNTP: 1,"2023/05/15,05:09:31"
```
It seems to have ignored the timezone (-28) and reported UTC time.