Xinyuan-LilyGO/LilyGO-T-SIM7000G #235: Confusion about RX/TX pins

morgana2313, 2023-05-22:


trentp-igor, 2023-05-25:
These pins need to be crossed-over when connected from the SIM7000 to the ESP32.  I.e., TX on the SIM7000 needs to be connected to the RX pin on the ESP32.  The label pin 26 RX is from the perspective of the SIM7000.  On the ESP32, this is the TX pin.

So one could label them either way, depending on which device's perspective is used.  It's good to be consistent, otherwise it's easy to get things wrong.

Here, we see the first pin supplied in the constructor is named the TX pin, which is pin 26 in this file:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/be1b3015ca89888afd5731108275c8e791b4259a/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino#L234

But here the first pin supplied named the RX pin, which is also pin 26:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/be1b3015ca89888afd5731108275c8e791b4259a/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino#LL109C5-L109C5

So in both cases pin 26 is the first one.  It's just named from the ESP32 perspective in the former and the SIM7000 in the latter.

xyzzy42, 2023-05-25:
These pins need to be crossed-over when connected from the SIM7000 to the ESP32.  I.e., TX on the SIM7000 needs to be connected to the RX pin on the ESP32.  The label pin 26 RX is from the perspective of the SIM7000.  On the ESP32, this is the TX pin.

So one could label them either way, depending on which device's perspective is used.  It's good to be consistent, otherwise it's easy to get things wrong.

Here, we see the first pin supplied in the constructor is named the TX pin, which is pin 26 in this file:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/be1b3015ca89888afd5731108275c8e791b4259a/examples/Arduino_Azure_IoTHub/Arduino_Azure_IoTHub.ino#L234

But here the first pin supplied named the RX pin, which is also pin 26:
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/be1b3015ca89888afd5731108275c8e791b4259a/examples/Arduino_NetworkTest/Arduino_NetworkTest.ino#LL109C5-L109C5

So in both cases pin 26 is the first one.  It's just named from the ESP32 perspective in the former and the SIM7000 in the latter.

morgana2313, 2023-05-25:
I think in thus user documentation the pin names  from the SIM7000's perspective is not very relevant; the user most likely thinks from the ESP32 perspective. :-)

At least this was my perspective when configuring esp-idf menu config and was asked for the (ESP) RX and TX GPIO numbers.

tony-png, 2024-03-03:
THANK YOU!

Ok guys, this one drove me CRAZY... please correct the diagram. It makes no sense to list them the other way around!!!!

Correct pins are RX = 26 TX = 27!

tony-png, 2024-03-03:
Code for micropython: 

`# Configuration for the OLED display using SSD1306 driver

# I2C Pin configuration
I2C_SCL_PIN = 22  # SCL pin
I2C_SDA_PIN = 21  # SDA pin

# OLED Display I2C address
OLED_I2C_ADDR = 0x3C

# BNO055 I2C address
BNO055_I2C_ADDR = 0x28

# OLED Display dimensions
OLED_WIDTH = 128
OLED_HEIGHT = 64
WIFI_SSID = 'NOKIA-B8F2'  # Replace with your WiFi SSID
WIFI_PASSWORD = 'wGaP5Lz6ka'  # Replace with your WiFi password

# GPS Module Pin configuration for ESP32 with SIM-7000G
GPS_TX_PIN = 27  # TX pin
GPS_RX_PIN = 26  # RX pin
# UART configuration
UART_BAUD = 115200
GPS_PWR_PIN = 4
GPS_LED_PIN = 12

from machine import UART, Pin
import time

# Initialize the power pin
power_pin = Pin(GPS_PWR_PIN, Pin.OUT)

# Function to power cycle the modem
def power_on_modem():
    print('powering ON modem...')
    # Turn on the modem
    power_pin.value(1)
    # Wait a bit for the modem to initialize
    time.sleep(5)
    # Set GPS LED pin to high
    led_pin = Pin(GPS_LED_PIN, Pin.OUT)
    led_pin.value(0)

def power_off_modem():
    print('powering off modem')
    power_pin.value(0)
    time.sleep(5)
    
# Power cycle the modem
power_off_modem()
power_on_modem()

# Initialize UART with the GPS module
uart = UART(1, baudrate=UART_BAUD, tx=Pin(GPS_TX_PIN), rx=Pin(GPS_RX_PIN))

# Now you can proceed with sending commands or data to the GPS module
# For example, sending a test command
# Attempt to send AT command up to 10 times to ensure the module gets the baud rate
for attempt in range(10):
    uart.write('AT\r\n')
    # Give the module some time to respond
    time.sleep(1)
    # Read the response
    response = uart.read()
    if response:
        print("Received response:", response)
        break  # Exit the loop if a response is received
    else:
        print(f"No response on attempt {attempt + 1} - retrying")

if not response:
    print("Failed to communicate with the module after 10 attempts - check UART initialization and wiring")

`

iomari, 2024-03-22:
I'm confused. Can someone tell me exactly which pins on my esp32 t-sim7000g device should I use to connect the sim module?
On the diagram in the lower left corner, I see sim7000g pins 26 and 27. Is that the same as the pins gpio25 and gpio27? does that mean that I connect these pins to the corresponding pins labeled TXD and RXD? I'm basing all this on the diagram:
 https://i0.wp.com/randomnerdtutorials.com/wp-content/uploads/2022/08/LILYGO-SIM7000G-V1_1-ESP32-pinout.jpg?w=1000&quality=100&strip=all&ssl=1

I want to know exactly what connect to what based on that diagram so that I can start coding to send sms messages. Also on Where do I connect the POWER 4 pin on the bottom left also.