Xinyuan-LilyGO/LilyGO-T-SIM7000G #237: Battery and 5V pin

cicarelli, 2023-05-28:
I have an external 5V power source, but I need to use the 18650 battery for emergencies. Does I need to use the “solar panel interface” to power the module, or can I use the 5V pin? Or, in other words... does the battery remains charged if I power the module using the 5V pin?

xyzzy42, 2023-06-05:
The 5V pin goes to same place as the USB-C power, which will charge the battery.  This is the better way to use your 5V source than the solar in pin.

However, you shouldn't power it over the 5V pin and use USB-C at the same time.  Using the solar in at the same time as USB-C/5V is ok, as the USB-C power effectively turns off the solar.

5V will also power the CP2104 USB to UART chip's vbus line, which solar in or battery does not do.

The 5V will power the system "directly", via a switching regulator to 4.2 V and then a linear regulator down to 3.3.  The solar will only go the battery charger circuit and charge the battery.  And then the battery will power a different switching regulator to get stepped up to 4.2 V and then to the same linear regulator back down to 3.3V.

So the solar input will less efficiently charge the battery first and goes through more up/down regulator steps.

cicarelli, 2023-06-06:
This is great for my project. Thank you for your help, @xyzzy42 !

cicarelli, 2023-06-06:
I think I can close this issue. Thanks.

closed 2023-06-06 by cicarelli