Xinyuan-LilyGO/LilyGO-T-SIM7000G #239: Removal of on\ off SPDT

glen020, 2023-06-19:
what is the wiring look like for the onboard on\off switch?

looks to be a SPDT, does it perform any other function other than using the battery? is there anything connected to the other poles?  i want to change it to a self latching water resistant switch and put it in a ip67 case. Is this doable? or is there logic im missing

xyzzy42, 2023-06-27:
It's in the schematic.  There's nothing on the other pole.  The switch connects the battery to the power supply circuit.  The charger and protection circuit is on the same side of the switch as the battery.  So it will still charge when the switch is off.

glen020, 2023-10-31:
Thanks, If you buy enough they will remove from the factory as well.

closed 2023-10-31 by glen020