Xinyuan-LilyGO/LilyGO-T-SIM7000G #240: Using the board as 4G and WiFi router

gittimotap, 2023-06-20:
Hi

Is it possible to use the board as a WiFi AP to provide connectivity to other devices, but use 4G on the board for the connection for the board itself?
Lilygo->4G->Internet
Devices->WiFi ->Lilygo->Internet

If yes, do you have any code samples?

DevinCarpenter, 2023-07-30:
This is not a high speed 4G internet board. The Sim7000 is an IOT board for low speed reliable data. Use the Sim7600G or other similar modem to achieve anything faster than 300kbps.

realbeeit, 2024-02-25:
I am interested also in same thing. I do not need fast internet I need reliable connection. Not for mobile devices or computers, but for other IoT devices eg. other ESP32 module.

DevinCarpenter, 2024-03-06:
> I am interested also in same thing. I do not need fast internet I need reliable connection. Not for mobile devices or computers, but for other IoT devices eg. other ESP32 module.

For sending data in general - yes this board will work. It is made for sending IoT device data. The board provides a stable and reliable connection but at slow speeds - 300Kbps(DL)/375Kbps(UL). See the spec for reference:  https://www.simcom.com/product/SIM7000X.html

Kagee, 2024-04-17:
I am also interested, spesifically if there is any ready code or project that will make the ESP work as a NAT router and forward traffic from other IOT devices connected to a esp32 AP to the internet through a SIM module, be it SIM7000, SIM7006 or other. Speed is not required.

DevinCarpenter, 2024-07-02:
@Kagee not that I'm aware of. I write the code myself if I need specific features. You may be able to find examples that get you close to what you need.