Xinyuan-LilyGO/LilyGO-T-SIM7000G #244: Internal battery charging

printrmd, 2023-09-10:
Hello
Is it possible to charge the internal battery not through the USB input 
because when there is a power fail and there is no voltage to charge it, 
the modem reboots or when the power returns.
Thanks.

realbeeit, 2023-10-06:
You can charge it through solar input.

printrmd, 2023-10-06:
hi
Can I connect a power supply or battery to the input of a solar panel, 
what voltage should it be
Thanks
From: BeeIT 
Sent: Friday, October 6, 2023 11:24 AM
To: Xinyuan-LilyGO/LilyGO-T-SIM7000G 
Cc: printrmd ; Author 
Subject: Re: [Xinyuan-LilyGO/LilyGO-T-SIM7000G] Internal battery charging (Issue #244)

You can charge it through solar input.

realbeeit, 2023-10-06:
Solar input is anywhere between 0 and 6V.

trentp-igor, 2023-10-31:
The charger used has a minimum voltage of 4.35 V.  So the input needs to be 4.35 V - 6 V in order to charge.  Under 4.35 V won't break anything, but it won't charge.  At least according to the datasheet of the CN4065.  It doesn't have a boost converter, so the input voltage needs to be greater than the battery's voltage.

If the battery is discharged down to 3V, then maybe it can charge on the a 3.1 V input?  At least until the battery voltage rises to meet the input voltage.  Datasheet seems to indicate 4.35 V (or sometimes 4.4 V, it's inconsistent) is the minimum, but I wonder if that's really the case.

xyzzy42, 2023-10-31:
The charger used has a minimum voltage of 4.35 V. So the input needs to be 4.35 V - 6 V in order to charge. Under 4.35 V won't break anything, but it won't charge. At least according to the datasheet of the CN4065. It doesn't have a boost converter, so the input voltage needs to be greater than the battery's voltage.

If the battery is discharged down to 3V, then maybe it can charge on the a 3.1 V input? At least until the battery voltage rises to meet the input voltage. Datasheet seems to indicate 4.35 V (or sometimes 4.4 V, it's inconsistent) is the minimum, but I wonder if that's really the case.

billsargent, 2024-01-22:
When plugged in via USB, does this actually charge the battery? The documentation isn't clear on that.

printrmd, 2024-01-23:
Hi
i have new esp32 with A7670G there is two usb connector way?

thanks

From: billsargent 
Sent: Tuesday, January 23, 2024 1:25 AM
To: Xinyuan-LilyGO/LilyGO-T-SIM7000G 
Cc: printrmd ; Author 
Subject: Re: [Xinyuan-LilyGO/LilyGO-T-SIM7000G] Internal battery charging (Issue #244)

When plugged in via USB, does this actually charge the battery? The documentation isn't clear on that.

billsargent, 2024-01-23:
I checked with my meter last night and there is voltage from USB to the 18650 holder on the back. So it apparently does charge by USB without the need for a solar panel or some external power sourcce connected to the solar input.