Xinyuan-LilyGO/LilyGO-T-SIM7000G #246: SIM7000G positioning and cellular

BestITUserEUW, 2023-09-28:
Hey,
I wanted to know if it's possible to use positioning and cellular at the same time with SIM7000G ? You only mention SIM7070 not supporting that, but I couldn't find examples on position and cellular at the same time. I'm not too familiar with modems but from what I read here:
https://github.com/espressif/esp-protocols/blob/master/docs/esp_modem/en/README.rst
 
You have different modes your modem can be in DATA for cellular and COMMAND for sending and receiving at commands. There is also CMUX which can enable both, in the readme it's said that sim7000 series does not support cmux at all. I can't use TinyGsm because there is no support for x509 Certficate authentication which I need for my Mqtt broker, so I am using esp_modem from esp-idf with idf framework right now. I was able to connect to the broker, publish and receive data with ssl connection and also switch between data and command mode and thus publishing Gps data, but I wasn't able to send at commands while having cellular connection open. 

Is it possible that I can get gps data and publish over cellular at the same time ?

BestITUserEUW, 2023-12-06:
@Jorin-Post 
My Problem is that i want to get Gps data from device and publish over lte at the same time and as far as i know, for that you have to switch between data and command mode in modem.

For your problem, I can't really send you the code it's not open source. I haven't found a way to get ssl handshake done with Arduino Framework and LTE, but for esp-idf framework i found a solution using PPP (Point to Point Protocol) Client. With that you can establish ssl connection with LTE. Here is an example for pppos client in esp-idf. 

https://github.com/espressif/espprotocols/blob/master/components/esp_modem/examples/pppos_client/main/pppos_client_main.c

You will also need esp_modem library and esp_mqtt library from esp-idf. In esp_mqtt client setup you can add your ssl certs:

> esp_mqtt_client_config_t mqttConf{};
> mqttConf.broker.verification.certificate = "root_cert";
> mqttConf.credentials.authentication.certificate = "client_cert";
> mqttConf.credentials.authentication.key ="client_key";

If you don't need modem and you are publishing over Wifi just use WiFiClientSecure like so:

> WiFiClientSecure wcs{};
> wcs.setCACert("ca_cert_data");
> wcs.setCertificate("certificate_data");
> wcs.setPrivateKey("private_key_data");

Then pass wcs to your mqtt library on initialization. For example 256dpi/MQTT:

> MQTTClient mqttClient{};
> mqttClient.begin("myEndpoint", 8883, wcs);

Jorin-Post, 2023-12-06:
Esp IDF is a bit more advanced than what I used to. But thx I'm going to try it. For your problem I have unfortunately no fix.