Xinyuan-LilyGO/LilyGO-T-SIM7000G #251: Modem will not power off

nravanelli, 2023-11-08:
I have been attempting to power down the modem using the commands provided, and methods presented in various issues here, but have not had success.

my setup runs this function to start the modem:
```
  pinMode(MODEM_PWKEY, OUTPUT);
  digitalWrite(MODEM_PWKEY, HIGH);
  // Starting the machine requires at least 1 second of low level, and with a level conversion, the levels are opposite
  delay(1000);
  digitalWrite(MODEM_PWKEY, LOW);
```

and then I initially turn the modem on:

```
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(10);
  digitalWrite(MODEM_PWKEY, LOW);
  delay(1010);  //Ton 1sec
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(4510);  //Ton uart 4.5sec but seems to need ~7sec after hard (button) reset
  SerialAT.begin(115200, SERIAL_8N1, MODEM_TX, MODEM_RX);
  modem.setBaud(115200);
  modem.begin();
```
and then I turn off modem 
```
  modem.sendAT("+CPOWD=1");
    if (modem.waitResponse(10000L) != 1) {
        Serial.println("+CPOWD=1");
    }
  modem.poweroff();
  digitalWrite(MODEM_PWKEY, HIGH);
  delay(6000);//Ensure that SIM7000 is powered off after a long delay
```

But, after a couple seconds of running my modem off function, the modem turns back on (the network leds flash). 

Is there a foolproof solution to turning the modem off and still run code on the ESP32?

felmue, 2023-11-12:
Hello @nravanelli 

from the schematic I see there is an NPN transistor used for the PWR_KEY which means the levels are inverted.

So to turn on the modem GPIO4 should be driven LOW/HIGH/LOW which after the transistor is HIGH/LOW/HIGH.

And since, after sending the AT command to turn off, you set MODEM_PWKEY to HIGH this actually makes the modem turning on again. Which is what you see.

BTW: this comment (from your post) actually tries to tell you that the PWR_KEY line is inverted. (opposite = inverted)

_// Starting the machine requires at least 1 second of low level, and with a level conversion, the levels are opposite_

Thanks
Felix

nravanelli, 2023-12-04:
@felmue Thanks Felix, I solved based on your comment. Is there a way to turn off the modem at the start, without having to turn it on first?