Xinyuan-LilyGO/LilyGO-T-SIM7000G #254: Compatibility of SIM7000G: Simultaneous Use of Positioning and Cellular Connection

mrcl1234, 2023-12-26:
Hello community,

I am currently working with the SIM7000G module and I've come across information in the official documentation that states the SIM7070 positioning and cellular connection cannot be used simultaneously.

I would like to know if a similar limitation applies to the SIM7000G. Can someone provide insights into whether it's possible to use positioning and cellular connection at the same time with the SIM7000G module? Your experiences or any relevant information would be greatly appreciated.

Thank you!

ZoiTho, 2024-11-06:
yes you can by restarting modem, enabling gps, get gps data, then disable gps and enable gprs connectivity and then doing this all over again. Simultaneoulsy no.

lewisxhe, 2024-11-07:
The document description refers to SIM7070G, not SIM7000G

ZoiTho, 2024-11-07:
Indeed the document description refers to SIM7070G .
 In LilyGO boards such as ESP32 WROVER and SIM7000G , the GPS pins that could connect simcom chip's UART3 to esp32 are not connected so you have to find a way to connect these SIM7000G pins to your ESP32 board. But you have to do it by yourself and take into account the differences of ESP32 and SIM7000G chips .