Xinyuan-LilyGO/LilyGO-T-SIM7000G #256: Enabling WiFi prevents SD card operations

henriberisha, 2024-01-13:
Hello, I am using a 8GB SDHC card. The issue i am having is related to how WiFi enabling will prevent reading from the SD card. I am not sure what the reason is. but i put together a simple sketch to demonstrate the issue.

Thank you!

```#include <WiFi.h>
#include <WiFiClientSecure.h>
#include <PubSubClient.h>
#include "FS.h"
#include "SD.h"
#include "SPI.h"

// LilyGO T-SIM7000G Pinout
#define UART_BAUD 115200
#define PIN_DTR 25
#define PIN_TX 27
#define PIN_RX 26

#define SD_MISO 2
#define SD_MOSI 15
#define SD_SCLK 14
#define SD_CS 13

#define PWR_PIN 4
#define LED_PIN 12

//  Your WiFi credentials
const char *ssid = "your_ssid";
const char *password = "your_password";

void readFile(fs::FS &fs, const char *path)
{
    Serial.printf("Reading file: %s\r\n", path);

    File file = fs.open(path);
    if (!file || file.isDirectory())
    {
        Serial.println("- failed to open file for reading");
        return;
    }

    Serial.println("- read from file:");
    while (file.available())
    {
        Serial.write(file.read());
    }
    file.close();
}

void writeFile(fs::FS &fs, const char *path, const char *message)
{
    Serial.printf("Writing file: %s\r\n", path);

    File file = fs.open(path, FILE_WRITE);
    if (!file)
    {
        Serial.println("- failed to open file for writing");
        return;
    }
    if (file.print(message))
    {
        Serial.println("- file written");
    }
    else
    {
        Serial.println("- write failed");
    }
    file.close();
}

void WiFiStationConnected(WiFiEvent_t event, WiFiEventInfo_t info)
{
    Serial.println("Connected to AP successfully!");
}

void WiFiGotIP(WiFiEvent_t event, WiFiEventInfo_t info)
{
    Serial.print("WiFi connected with IP address: ");
    Serial.println(WiFi.localIP());
}

// This is a callback function and will be called async at anypoint in the program if the wifi is not connected
void WiFiStationDisconnected(WiFiEvent_t event, WiFiEventInfo_t info)
{
    Serial.print("Not connected to a WiFi Access Point. Reason: ");
    Serial.println(info.wifi_sta_disconnected.reason);
}

void connectWiFi()
{
    // Start WiFi connection
    WiFi.mode(WIFI_STA);
    WiFi.config(INADDR_NONE, INADDR_NONE, INADDR_NONE, INADDR_NONE);
    WiFi.setHostname("myESP32"); // define hostname
    WiFi.begin(ssid, password);
    Serial.print("Connecting to WiFi ..");
    while (WiFi.status() != WL_CONNECTED)
    {
        Serial.print(".");
        delay(1000);
    }
}

void setup()
{
    Serial.begin(115200);
    delay(3000);

    /* initialize SD card */
    SPI.begin(SD_SCLK, SD_MISO, SD_MOSI);
    if (!SD.begin(SD_CS))
    {
        Serial.println("SD Card MOUNT FAIL");
        return;
    }

    uint8_t cardType = SD.cardType();

    if (cardType == CARD_NONE)
    {
        Serial.println("No SD card attached");
        return;
    }

    Serial.print("SD Card Type: ");
    if (cardType == CARD_MMC)
    {
        Serial.println("MMC");
    }
    else if (cardType == CARD_SD)
    {
        Serial.println("SDSC");
    }
    else if (cardType == CARD_SDHC)
    {
        Serial.println("SDHC");
    }
    else
    {
        Serial.println("UNKNOWN");
    }

    int64_t cardSize = SD.cardSize() / (1024 * 1024);
    Serial.printf("SD Card Size: %lluMB\n", cardSize);

    // Set LED OFF
    pinMode(LED_PIN, OUTPUT);
    digitalWrite(LED_PIN, LOW);

    writeFile(SD, "/myTest.txt", "Testing File\n\n");

    readFile(SD, "/myTest.txt");

    WiFi.disconnect(true);

    readFile(SD, "/myTest.txt");

    WiFi.onEvent(WiFiStationConnected, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_CONNECTED);
    WiFi.onEvent(WiFiGotIP, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_GOT_IP);
    WiFi.onEvent(WiFiStationDisconnected, WiFiEvent_t::ARDUINO_EVENT_WIFI_STA_DISCONNECTED);

    readFile(SD, "/myTest.txt");

    connectWiFi();

    readFile(SD, "/myTest.txt");

    WiFi.disconnect(true, true);
}

void loop()
{

    readFile(SD, "/myTest.txt");
    delay(3000);
}

/////////////////////////////////////////////////////////////////////////////////////////////////```
![image](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/assets/99278453/2875a2f1-a829-4643-835c-6f8b01f314c1)

henriberisha, 2024-01-13:
[image]

billsargent, 2024-01-23:
Here's what I get after it connecfts to Wifi.

`- read from file:
Testing File

Reading file: /myTest.txt
- read from file:
Testing File

Reading file: /myTest.txt
- read from file:
Testing File`

papsngbayan, 2025-02-14:
Also encountered this on my 7070G, any updates?