Xinyuan-LilyGO/LilyGO-T-SIM7000G #275: SIM7000G ok but SIM7070G drops data on LTE-M TCP connections

baldvinhansson, 2024-07-06:
Dear all.

I have code connecting to the network using LTE-M. I program the ESP32 using Arduino IDE.

Defining my parameters for connecting the modem I use 

```
#define UART_BAUD           9600
#ifdef TINY_GSM_MODEM_SIM7070
	#define PIN_DTR             32
#else
	#define PIN_DTR             25
#endif
#define PIN_TX              27
#define PIN_RX              26
#define PWR_PIN             4

```
Then setting up the network I use

```
modem.setNetworkMode(38); // LTE only
modem.setPreferredMode(1); // CAT-M

```

Using TinyGSM with RX buffer of 1024 bytes

`#define TINY_GSM_RX_BUFFER 1024 // Set RX buffer to 1Kb
`
Then connecting to a TCP server using 

`TinyGsmClient _tcpClient(modem, 0);
`

Then it will send data packets of various sizes, 20-100 bytes each time and receive response of 2-80 bytes each time.

The problem is that the SIM7000G works flawlessly. It can connect and communicate for days, no problem, ever. The SIM7070G however is unable to communicate without dropping data for any more than (randomly) 3-6 interactions/transactions with the server before data coming from the server to the SIM7070G is lost somewhere in transit.

It is a bit random exactly how many transactions it can do before stopping, but never more than a few.

I have looked through datasheets, schematics (though I'm still not able to find one for the T-SIM7070G) and libraries and I am unable to pinpoint a reasonable explanation as to why the SIM7070G would behave like this. Any hints or help is greatly appreciated.

lewisxhe, 2024-08-17:
SIM7070 has fewer TCP commands than SIM7000. Using TinyGSM directly may not work well. It is recommended to use AT commands directly.
For how to use, please refer to the manual https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/docs/SIM7070/SIM7070_SIM7080_SIM7090%20Series_TCPUDP(S)_Application%20Note_V1.03.pdf

[image]

lewisxhe, 2024-08-17:
There are still differences between SIM7070 and SIM7000 TCP commands. Please refer to the application manual for data transmission test. https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/docs/SIM7070/SIM7070_SIM7080_SIM7090%20Series_TCPUDP(S)_Application%20Note_V1.03.pdf