Xinyuan-LilyGO/LilyGO-T-SIM7000G #277: SIM7070G Modem Power-on Failure

tadenham, 2024-07-13:
I have several weather stations deployed with SIM7070G boards and all have had this same issue at some point. These weather stations transmit regularly at 15-minute intervals, powering on the modem and then shutting down again when the transmission is complete. They will all operate normally anywhere from several days to several months, then suddenly the modem will fail to start. This can be verified by the STATUS and NETLIGHT LEDs not turning on when the modem is supposed to be turned on. This drains the battery very quickly as the ESP32 will wait from a response from the modem even though it is not powered on, preventing the board from going into deep sleep for a while. I can also confirm that the ESP32 still functions normally throughout this as it continues to write the data to the SD card when the transmission is unsuccessful.

The boards have never recovered from this condition, the modem will fail to power on indefinitely after this point. The modem will only operate normally once again when battery power is completely removed and the board is rebooted. This has become a major issue as these stations are located in difficult to access remote locations.

I have a few other stations with SIM7600 boards and they have operated for nearly a year so far without issue.

baldvinhansson, 2024-07-24:
I'm sorry I don't have a solution to your problem, but want to ask if you've been able to use SIM7070G for transferring data over TCP connected to an LTE network with any consistency and reliability? And if so, would you be willing to share any or all of the relevant code?

The reason I ask is that I've not been able to do this at all and I've raised an issue here on github for this a while back and emailed LILYGO and I get no replies or answers anywhere. I'm able to do this fine on SIM7000G but SIM7070G starts out ok but after a few seconds it simply will not receive any data from the remote end/server anymore and the connection stops.

Thank you for considering this and sorry I don't have a solution at the moment to your original issue.

tadenham, 2024-07-24:
Although I now use UDP to send data, I used to use MQTT which uses TCP. I didn't have any issues other the occasional dropped connection due to poor service, but I switched to UDP because of the lower overhead. I also added an acknowlegment from the server so that all data has guaranteed delivery.

I haven't used the SIM7000G so can't say if they behave any differently but as far as I know the AT command set is the same. I don't use any external libraries since I write the code exclusively in ESP-IDF. You might not find the following useful if you're using the libraries, but the code for both MQTT and UDP connections is in there. Much of the code is weather station stuff unrelated to the actual transmission of data.

[https://github.com/tadenham/Cellular-AWS](https://github.com/tadenham/Cellular-AWS)

baldvinhansson, 2024-07-24:
Thank you for the prompt and useful answer. I will keep an eye out for potential solutions to the issue you originally described.

tadenham, 2025-06-16:
> Hola [@tadenham](https://github.com/tadenham) , pudiste resolver el problema del modem del SIM7070G? estoy interesado en hacer algo similar, pero buscando información, pude encontrar tu problema, que realmente es un problema importante para estacoines remotas.
> 
> Pudiste hacer la misma prueba con módulos separados? es decir, un ESP32 + SIM800L + NEO-6M + Modulo SD + MCP73871 ?

I was never able to find a solution. I've purchased several of these boards over the past few years. Some of them work perfectly forever, many others fail in a matter of days. It seems to me that it's an issue of defective or poor quality components for the modem power supply, which could explain why some boards are fine and others are not. LilyGo has been of no assistance, so I found it easier just to design my own PCBs to fit my application rather than deal with these issues.

germanh1980, 2025-06-16:
Qué pena que sea así, ya que es una solución bastante práctica. Y me alegro que hayas podido salir del problema.