Xinyuan-LilyGO/LilyGO-T-SIM7000G #283: Different kinds of LTE sessions ?

tronar, 2024-10-16:
Hi, I'm struggling to understand the different kinds of connections that a device can mantain. 
I'm programming a T-SIM7000G and kind of understand that in LTE you may have more than one PDP context with different profiles.
I also understand that the modem supports high level protocol sessions that use some form of high level session object. 
But I have not been able to clearly understand which is related to what:

AT+CGATT (network attachment aka registration?)
AT+CGACT (PDP session ?)
AT+CNACT 
AT+SAPBR 

And I'm leaving high level ones like IP/MQTT/HTTP. 
Is there any document that describes these ?

deeja, 2025-06-10:
There's a bunch of manuals that you can look at, but start here:
https://cdn.geekfactory.mx/sim7000g/SIM7000%20Series_TCPIP_Application%20Note_V1.02.pdf
And for the details on each AT command:
https://cdn.geekfactory.mx/sim7000g/SIM7000%20Series_AT%20Command%20Manual_V1.06.pdf

If you are using TinyGsm, then it does a whole lot of commands inside each method, and sometimes they're not the ones you want.

tronar, 2025-06-10:
Thank you for your attention. No, I'm not using TinyGSM, just plain AT commands and the sessions I was referring to are not conceptually defined in those manuals. In fact, the manuals count on you knowing what you are trying to do and what all parameters mean, which is odd as no manual defines those.

deeja, 2025-06-11:
I think the application note manual should help. 
https://cdn.geekfactory.mx/sim7000g/SIM7000%20Series_TCPIP_Application%20Note_V1.02.pdf
There's a step by step for the various tcp/udp connections, and the multiple PDP contexts like you mentioned.

tronar, 2025-06-11:
Deeja,
I do have the code working, but I do not understand what are the dependencies involved, as then I do not understand what is needed to recover from some situations.
Some of the "sessions" depend on having some prerequisites met. Like to have a PDP context you need to be registered.
All the notes I've read do not explain the relation. They just explain each command and some do show some example use cases at the end. I was hoping for a document that explains the high level picture, or for someone to be kind enough to explain it.

BTW, I've read the TCP app note (I'm not using TCP connections, letting the module to do it) and the MQTT app note (which I'm using) and the NTP app note, and the SSL app note, and...

deeja, 2025-06-12:
Ok, no problem. Hopefully someone can help you with that