Xinyuan-LilyGO/LilyGO-T-SIM7000G #286: SIMCOM support is non existent, Lilygo here is near 0, still...

tronar, 2024-10-30:
Does anybody know if there is some way of getting remote RSSI indication ?
I.e. the local RSSI reflects the DL conditions, but how can I get an idea of UL consitions ?

TIA

tronar, 2024-10-31:
Radio link is between UE and eNB. 
The "local" RSSI from the UE point of view (the SIM7000) is the Received Signal Strength Indicator that relates to the DL (downwards link).
The "remote" RSSI would be the signal strength as seen from the eNB, i.e, related to the UL (uplink).
From a protocol point of view, the eNB is in charge of the control of the link, and the UE reports signal conditions so the eNB can adjust. But it also has a way of telling the UE to tx at higher or lower power level. Do we have any way of seeing such information ?

lewisxhe, 2024-11-01:
The module can only obtain the RSSI value of the downlink, use AT+CPSI to query

tronar, 2024-11-01:
Too bad. Local RSSI is shown in more than one place. AT+CENG also shows Tx power but it is only valid when transmitting and I have not been able to latch it. Tx power is commanded by eNB and should follow remote conditions.
On the other hand, the info is for sure in the module so SIMCOM might have some undocumented mode to access it. SIMCOM documents are way below average, and as teh subject says, its support is to be found.

tronar, 2024-12-25:
Could you please ask SIMCOM for a way to read the TX power measure ?

lewisxhe, 2024-12-26:
It can be tested through cmw500. No commands are required during signaling test, and it can be sent through the instrument.

tronar, 2024-12-26:
I'm not sure I understand what you mean by this... I'm not trying to measure this in a test environment, I want to have a means to know the current TX power that should be the consequence of the tower (eNB) commanding it.

tronar, 2024-12-26:
Can you please ask SIMCOM of a procedure to be able to read tx power with AT+CENG ?
From the manual:
  <tx power> Tx power value in 1/10 dBm. <tx power> is only meaningful
  when the device is in traffic. When there is no traffic, the value
  is invalid.The value of <tx power> is 255.
but I have been unable to read actual TX power, I always get 255.

lewisxhe, 2024-12-30:
The following is the test command replied by SIMCOM, you can test it
[image]

lewisxhe, 2024-12-30:
Added CTBURST_Application_Note

tronar, 2024-12-30:
Lewisxhe,
That command family seems not for this product. I tried no matter what:

ATI
SIM7000G R1529

OK
AT^BODYSARON=1
ERROR

Also, I read the CTBURST application note, and it is intended for testing, assuming it works for the SIM7000 family which does not seem to be the case:

at+CFUN=5
OK
AT+CTBURST=?
ERROR
AT+CTBURST=0
ERROR

Just in case, please take a minute to understand my need. I need to read during working conditions the current TX power level of the modem. According to the LTE norm, this level is determined by the base station (eNB) depending on the uplink conditions, thus is a fair estimator of the uplink conditions. I have some areas where the node is failing to connect and in field testing  this information would be most valuable.

Thank you!

lewisxhe, 2024-12-31:
Your requirement is to read the current transmit power level of the modem at runtime. SIMCOM replied that there is no such function. Sorry.

tronar, 2024-12-31:
Ok, thank you. It seems whoever replies just want to reply something, given what I already stated about the AT+CENG command.
From the SIM7000 AT manual:

Engineering Mode is designed to allow a field engineer to view and test the
network information received by a handset, when the handset is either in
idle mode or dedicated mode (that is: with a call active). In each mode, the
engineer is able to view network interaction for the "serving cell" (the cell the
handset is currently registered with) or for the neighboring cells.

If camping on a CAT-M or NB-IOT cell:
+CENG: <mode>,<Ncell>,<cell num>,<System Mode>
[+CENG:
<cell>,"<earfcn>,<pci>,<rsrp>,<rssi>,<rsrq>,<sinr>,<tac>,<cellid>,<mc
c>,<mnc>,<tx power>"

<tx power> Tx power value in 1/10 dBm. <tx power> is only meaningful
when the device is in traffic. When there is no traffic, the value
is invalid.The value of <tx power> is 255.

lewisxhe, 2025-01-02:
OK, I understand what you are trying to say.

I misunderstood..

So your real problem is that the TX power field returned by AT+CENG remains at 255, right?

lewisxhe, 2025-01-03:
SIMCOM requires you to provide the debugging log

How to connect USB and USB driver can be viewed [here](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/docs/How%20to%20update%20firmware.md)

The video of capturing log operation is here. The video demonstrates SIM7080 and the operation method is the same as SIM7000
[SIM7080GCaptureLog](https://www.youtube.com/watch?v=AEIriATGVb4&t=4s)

## Capture log 
[DiagGrabPro_V1.9.7z](https://drive.google.com/file/d/1-lnNPaYeiPiAmtlAELLkv4R3_I4lSNfO/view?usp=sharing)

After the connection log is captured, you can send AT+CENG to operate, and the log will capture your current network status, etc.

tronar, 2025-01-03:
Lewis,
the log of my doing what ? I'm asking how to do something and they ask what is going on ?
What do they want me to do, to have the logs of ? 
I fail to understand what's the value of a log given that I don't have a problem with a specific scenario.
They Should be able to see whatever in their own lab.

Still, I'm open to do whatever, just tell me what.

lewisxhe, 2025-01-03:
They require that we capture the modem operation log and send it to simcom for analysis.

tronar, 2025-01-03:
Lewis, this happens to a new modem in my bench, I'm trying to find out how to measure field status by programming the read of tx power, so I'm working with a test unit. Are you a technical person ? If so, you would understand that to analyse something there has to be something to be analysed first. I know what a log is, I have no problem in getting a log, but the device should be trying to do something during the log capturre for the log to have any value.
So what I'm asking is what should the device be doing during the log capture ?

Also , this is not the problem of one device AFAIK, I have more than one here. If they think that AT+CENG should send meaningfull data, it might make more sense to test something with another device ? In any case, please ask them to provide a script for the test they want.

lewisxhe, 2025-01-03:
Use your SIM card, register the network, send AT+CENG query, the modem will record these things, you can query several times, of course, according to the AT manual, TX POWER will only be displayed when there is data traffic, you also need to find a way to keep the modem connected to a server, SIMCOM can analyze whether the base station you are using supports this function. Remember to start capturing from the time the module is turned on, this is the reply they gave.

tronar, 2025-01-03:
The base station supports this function ? This is nonsense. You are telling me to send a log of anything. I will, but it would be much more efficient if this had any value... they are just keeping us busy.

tronar, 2025-01-04:
Here you are ... qmdl inside .zip
[20250104142709864.zip](https://github.com/user-attachments/files/18307154/20250104142709864.zip)

lewisxhe, 2025-01-06:
Ok thanks, I will send it to SIMCOM and see what they say

lewisxhe, 2025-01-07:
SIMCOM reply: It can only be queried within a short period of time when the modem TX is in action, otherwise it will immediately change to 255. Also, from your log, since there is no network access information, it is impossible to know whether you are using NB or CAT-M.
[image]

lewisxhe, 2025-01-08:
I think I understand the reason for the problem.
Simcom replied that their test method is as follows:
Use the SIM7000G USB port to dial PPP,
Then query on the AT port

If the data and query are both on the AT port, then the data has been sent by the time the query is made, and the power is 255.

tronar, 2025-01-08:
Then on a system like the LiyGO-T-SIM7000G, the feature is useless because all interaction with the modem is through the AT port.
It would be nice for a firmware update to change that behaviour, and latch the last power level but ...
Thanks.

tronar, 2025-01-22:
FTR, today I enabled NB-IOT on the modem and here... +CENG does return the TX power:
AT+CENG?
+CENG: 1,1,1,LTE NB-IOT

+CENG: 0,"9312,336,-62,-61,-3,20,35103,128281804,722,34,-130"

So it seems to be a firmware bug of sorts. Yes, it works for SIMCOM because they tested in NB, not in CAT-M