Xinyuan-LilyGO/LilyGO-T-SIM7000G #289: flow control

fgnievinski, 2024-11-28:
it seems the hardware flow control pins are not connected between sim7000 and esp32.

trying flow control (`AT+IFC=1,1`), it seems the modem never sends XOFF, only XON.

did anyone face a similar issue?

lewisxhe, 2024-11-28:
SIM7000G does not have flow control function

fgnievinski, 2024-11-28:
sorry, I mean software flow control, which seems supported, no?
[image]
[image]

lewisxhe, 2024-11-28:
OK, I haven't used this function, I need to ask simcom

lewisxhe, 2024-11-28:
You need to update the latest firmware, then use the log capture tool to capture the modem log, and then send it to simcom for analysis

Log capture is the same as SIM7080G, but you need to see how to connect USB here
[How upgrade fw](https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/docs/How%20to%20update%20firmware.md)
[Firmware 1529b10SIM7000G](https://drive.google.com/file/d/105Jix1EYtxPomX5H1_2VmJphmYDoCYPJ/view?usp=sharing)
[Video](https://www.youtube.com/watch?v=AEIriATGVb4)

fgnievinski, 2024-11-28:
thank you for the video!

I've found the log capture tool (SIMCOM DiagGrabPro) here:
https://momoiot.co.kr/lte-at/module/diagnostic/

thanks also for the newer modem firmware, I'm still using 1529B08.