Xinyuan-LilyGO/LilyGO-T-SIM7000G #293: Sim7000g LTE Cat-M1 support and cellular + gps simultaneous support

swebt, 2024-12-18:
Hi, hope all is well

We are working on development using the sim7000g we purchased. 

some questions:
On the GitHub main page it mentions “ SIM7000/SIM707G does not support 4G network. Please ensure that the USIM card you use supports 2G(GSM)/NB-IOT access,SIM7070 positioning and cellular cannot be used simultaneously. Please disconnect the network when positioning, and turn off GPS when connecting to the network”

1. Does this mean that the Sim7000g does not work with LTE CAT-M1 state? The Sim7000 chip I think should support this but I’m not sure if this module supports that state?
2. It says the Sim7070 cannot use positioning and cellular simultaneously. Does this also apply to the Sim7000g?

thank you for the clarification.

lewisxhe, 2024-12-18:
SIM7000G only supports simultaneous GPS and data
SIM7070G does not support

swebt, 2024-12-19:
Great thanks!

sim7000g works with LTE Cat-M1?

lewisxhe, 2024-12-19:
[image]
Yes

swebt, 2024-12-27:
Great thank you!

closed 2024-12-27 by swebt