Xinyuan-LilyGO/LilyGO-T-SIM7000G #294: Unable to send AT commands

yallbinzapped, 2024-12-20:
Hi, can anyone help me figure out how to send AT commands to the t-sim7000g? I'm connected to the right port and I'm getting no response:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:4832
load:0x40078000,len:16460
load:0x40080400,len:4
load:0x40080404,len:3504
entry 0x400805cc
AT
+++AT
AT

lewisxhe, 2024-12-21:
Run this sketch
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/ATdebug

yallbinzapped, 2024-12-26:
Hi Lewis,

Thank you for the sketch. I've attached a log of the results.  Perhaps you can help me make sense of the AT Command test results:

Summary of Observations

  1.
Network Registration:
     *   The SIM card is successfully registered to the network (Rogers Wireless IoT) in roaming mode, as indicated by AT+CREG? returning +CREG: 0,5.
     *   The network detected is LTE (RAT: 7) as per AT+COPS? response.
  2.
Signal Quality:
     *   Signal quality: +CSQ: 31,99 The module has excellent signal strength, so connectivity issues are unlikely to be related to weak signals.
  3.
PDP Context Status:
     *   The PDP context configuration (via AT+CGDCONT?) indicates the APN is set to data.mono.
     *   Attempts to activate the PDP context (AT+CNACT=1,1) return OK, but the status remains +APP PDP: DEACTIVE.
  4.
Ping Results:
     *   Despite the PDP context showing as inactive, ICMP ping commands (AT+SNPING4="8.8.8.8",5,1,20000) to Google DNS (8.8.8.8) return results, albeit with high response times of 60000 ms.
  5.
Deactivation of PDP Context:
     *   The PDP context can be successfully deactivated (AT+CNACT=1,0 returns OK), but reactivation does not resolve the inactive status.

Sent from my Bell Samsung device over Canada’s largest network.

https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/tree/master/examples/Arduino_TinyGSM/ATdebug

Network registered.
Setting APN...
APN set successfully.
PDP context activated.
Network connected successfully.
Operator: Rogers Wireless IoT
Signal quality: 31
Modem ready. Type AT commands below:

Response: +APP PDP: ACTIVE

Command: AT
Response: OK

Command: AT+CMEE=2
Response: OK

Command: AT+CPIN?
Response: +CPIN: READY
Response: OK

Command: ATI
Response: SIM7000G R1529
Response: OK

Command: AT+CGMR
Response: Revision:1529B08SIM7000G
Response: OK

Command: AT+CIMI
Response: 295050950161982
Response: OK

Command: AT+COPS=0
Response: OK

Command: AT+COPS?
Response: +COPS: 0
Response: OK

Command: AT+CSQ
Response: +CSQ: 99,99
Response: OK

Response: +APP PDP: DEACTIVE
Response: DST: 0
Response: *PSUTTZ: 24/12/24,10:28:26","-20",0

Command: AT+CGACT=1,1
Response: OK

Command: AT+CSQ
Response: +CSQ: 31,99
Response: OK

Command: AT+CGPADDR
Response: +CGPADDR: 1,10.143.24.104
Response: +CGPADDR: 13,0.0.0.0
Response: OK

Command: AT+SNPING4="8.8.8.8",5,1,20000
Response: +SNPING4: 1,8.8.8.8,60000
Response: +SNPING4: 2,8.8.8.8,60000
Response: +SNPING4: 3,8.8.8.8,60000
Response: +SNPING4: 4,8.8.8.8,60000
Response: +SNPING4: 5,8.8.8.8,60000
Response: OK

Command: AT+CNACT=0,0
Response: +CME ERROR: invalid input value

Command: AT+CNACT=1,0
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CNACT=1,1
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CNACT=0,1
Response: +CME ERROR: invalid input value

Command: AT+CGACT=1,1
Response: OK

Command: AT+CGACT=1,0
Response: ERROR

Command: AT+CGACT=1,0
Response: ERROR

Command: AT+CGACT=1,1
Response: OK

Command: AT+CGACT=1,0
Response: ERROR

Command: AT+CGACT=1,1
Response: OK

Command: AT+CGACT=1,0
Response: ERROR

Command: AT+CNACT=1,0
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CGACT=1,1
Response: OK

Command: AT+CNACT=1,0
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CNACT=1,1
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CNACT=1,1
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CSQ
Response: +CSQ: 31,99
Response: OK

Command: AT+CNACT=1,1
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+SNPING4="8.8.8.8",5,1,20000
Response: +SNPING4: 1,8.8.8.8,60000
Response: +SNPING4: 2,8.8.8.8,60000
Response: +SNPING4: 3,8.8.8.8,60000
Response: +SNPING4: 4,8.8.8.8,60000
Response: +SNPING4: 5,8.8.8.8,60000
Response: OK

Command: AT+CNACT=1,0
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+SNPING4="8.8.8.8",5,1,20000
Response: +SNPING4: 1,8.8.8.8,60000
Response: +SNPING4: 2,8.8.8.8,60000
Response: +SNPING4: 3,8.8.8.8,60000
Response: +SNPING4: 4,8.8.8.8,60000
Response: +SNPING4: 5,8.8.8.8,60000
Response: OK

Command: AT+CSQ
Response: +CSQ: 31,99
Response: OK

Command: AT+CGDCONT?
Response: +CGDCONT: 1,"IP","data.mono","0.0.0.0",0,0,0,0
Response: +CGDCONT: 13,"IP","data.mono","0.0.0.0",0,0,0,0
Response: OK

Command: AT+CNACT=1,1
Response: OK

Response: +APP PDP: DEACTIVE

Command: AT+CGDCONT?
Response: +CGDCONT: 1,"IP","data.mono","0.0.0.0",0,0,0,0
Response: +CGDCONT: 13,"IP","data.mono","0.0.0.0",0,0,0,0
Response: OK

Command: AT+CREG?
Response: +CREG: 0,5
Response: OK

Command: AT+COPS?
Response: +COPS: 0,0,"Rogers Wireless IoT",7
Response: OK

lewisxhe, 2024-12-27:
Please test in the following order:
```bash
AT+CFUN=0
AT+CGDCONT=1,"IP","data.mono"
AT+CFUN=1
AT+CSQ
AT+CGATT?
AT+COPS?
AT+CGNAPN
AT+CNACT=1
AT+CNACT?
AT+SNPING4="www.google.com",3,16,1000
```
The above data.mono is assumed to be your APN. If not, please replace the correct APN settings