Xinyuan-LilyGO/LilyGO-T-SIM7000G #297: Over Voltage Problem

DiogoFerreira123321, 2025-03-10:
Hello i am writing this issure because i am using the lilygo7000G and i came across with something im not understanding.
the board has a overvoltage protection and an undervoltage protection.
I was doing some tests and the undervoltage works well, the board goes 0 when above the 2.4V more a less and when it gets higher the board goes to normal mode again but when i put more voltage, more than 4.3, 4.4V the board do not go to 0 V or another case is that when i turn the board with for exemple 4.7V and when the battery goes down the board does not wake up until i press the reset button or turn off and on the switch
can someone help me?

lewisxhe, 2025-03-10:
Please see the manual for the undervoltage and overvoltage protection IC DW02. There is no English version, only Chinese.

[DW02A.PDF](https://github.com/user-attachments/files/19159752/DW02A.PDF)

DiogoFerreira123321, 2025-03-10:
i have translated the document and i know the conditions of the overvoltage  but they are not working 🙁 the board is not waking up if its goes more than 4.275V and goes lower the 4.075V. neighter is cutting off then it goes above the 4.275V

Diogo Ferreira

www.atronia.eu<https://bit.ly/4fSYN7r>

***@***.******@***.***>

[cid:35513529-536c-4814-a6f6-ad46be6b1b3e]

Hardware

Developer

Rua Gil Vicente 174 R/C ESQ.,

Gafanha da Nazaré 3830-671

Ílhavo, Portugal

This message has been sent as a part of discussion between the sender and the addressee whose name is specified above. Should you receive this message by mistake, we would be most grateful if you informed us that the message has been sent to you. In this case, we also ask that you delete this message from your mailbox, and do not forward it or any part of it to anyone else. Thank you for your cooperation and understanding.

lewisxhe, 2025-03-10:
Where did you get the voltage from? How did you connect it for testing?

DiogoFerreira123321, 2025-03-10:
i have connected 2 supercapacitors to "be the battery" so i could do the hardware tests more efficiently.
ill explain you what i did! and thank you for the response 🙂

1- ‎charged the supercaps with 4.8V (to simulate a battery with more voltage that they can handle), then connected to the board,  the board was not on, i reduce the voltage of the supercapacitors ultil a voltage that the board should goes to normal state and the board wont go to normal state until i press manually the reset button of switch on off the switch.
2- when i have in the supercapacitors a normal working voltage, when i increase the voltage , instead of cutting the voltage it just goes 0.7 V lower , when i put 4.4 it goes 3.7 ,when i put 4.5 it goes 3.8 and it goes like this , im guessing the 0,7 drop voltage is because of the diode but it should cut off based on the datasheet right?

Diogo Ferreira

www.atronia.eu<https://bit.ly/4fSYN7r>

***@***.******@***.***>

[cid:7697150a-f855-4848-9ae6-296aafdcc1ae]

Hardware

Developer

Rua Gil Vicente 174 R/C ESQ.,

Gafanha da Nazaré 3830-671

Ílhavo, Portugal

This message has been sent as a part of discussion between the sender and the addressee whose name is specified above. Should you receive this message by mistake, we would be most grateful if you informed us that the message has been sent to you. In this case, we also ask that you delete this message from your mailbox, and do not forward it or any part of it to anyone else. Thank you for your cooperation and understanding.

lewisxhe, 2025-03-11:
Is this how it is tested? The overvoltage input must be connected from VIN, which will cut off the negative terminal of the battery

[image]

[image]

DiogoFerreira123321, 2025-03-11:
i am attaching a video of the simulation of a battery with overvoltage  and i discharge it and the board does not go to normal mode.

Diogo Ferreira

www.atronia.eu<https://bit.ly/4fSYN7r>

***@***.******@***.***>

[cid:89b165de-3ee0-430a-b8a7-ba591944d38a]

Hardware

Developer

Rua Gil Vicente 174 R/C ESQ.,

Gafanha da Nazaré 3830-671

Ílhavo, Portugal

This message has been sent as a part of discussion between the sender and the addressee whose name is specified above. Should you receive this message by mistake, we would be most grateful if you informed us that the message has been sent to you. In this case, we also ask that you delete this message from your mailbox, and do not forward it or any part of it to anyone else. Thank you for your cooperation and understanding.

image.png (view on web)<https://github.com/user-attachments/assets/44081d38-e499-48d2-95c6-4b201c171768>

image.png (view on web)<https://github.com/user-attachments/assets/42541e56-fd63-4e65-bddd-597a27fbf22e>

image.png (view on web)<https://github.com/user-attachments/assets/44081d38-e499-48d2-95c6-4b201c171768>

image.png (view on web)<https://github.com/user-attachments/assets/42541e56-fd63-4e65-bddd-597a27fbf22e>

DiogoFerreira123321, 2025-03-11:
i cant send the video the file is too big  is there someway i can send it so you guys can see?

lewisxhe, 2025-03-12:
You can upload to youtube