Xinyuan-LilyGO/LilyGO-T-SIM7000G #298: ESP32 to sim7000 uart serial inoperable

MikeyJohnston, 2025-04-11:
I can't get communication between the esp32 and the sim7000 to work. AT commands do not work, the commands from the TINYGSM and client do not work. At best they may work intermittently if they ever do. I can't connect to GPRS network. I have 1.8V on VDD_EXT and 3.3V on VDD_UP. 
I have updated the firmware on the sim7000 and have the SSCOM tool and have tested AT commands using that, and that works just fine. 
But the communication between the esp32 and sim7000 continues to fail. 
#define UART_BAUD_9600 9600
#define PIN_DTR 25
#define PIN_TX 27
#define PIN_RX 26
#define PWR_PIN 4
#define PIN_33 33 // Define the pin for IO 33
I have tried all different baud rates, I set the DTR pin high to keep the sim7000 awake, I've tried switching the RX and TX back and forth. 
I have tried a brand new sim7000G lilygo device with the default firmware, none of the AT commands work. I had 3 or 4 boards operating perfectly around the middle of March and now none of them work for the comms between esp32 and sim7000, I've reverted to allot of the examples on the lilygo website with no luck.

lewisxhe, 2025-04-14:
Did you change the baud rate?
Have you tested communication directly through the modem's USB AT port?

MikeyJohnston, 2025-04-14:
09:34:50.114 -> ......> It looks like the modem is not responding, trying to restart
09:35:59.433 -> ......> It looks like the modem is not responding, trying to restart
09:37:08.770 -> ......> It looks like the modem is not responding, trying to restart
09:38:18.095 -> ......> It looks like the modem is not responding, trying to restart
09:39:27.364 -> ......> It looks like the modem is not responding, trying to restart
09:40:36.700 -> 
It says this for every device but the modem lights are blinking. This is the Arduino_NetworkTest.ino from the github repo...
I have the 4pin USB for the sim7000 setup on 2 of the devices, and have updated the firmware using SSCom and the firmware 1529B08SIM7000G. Those two devices generate this loop of not responding. I also have a device that I have not soldered anything to, just the normal Lilygo SIM7000g... It is giving this response for this new device as well, I haven't updated firmware on that one yet. And I want to just keep that one un-touched/changed.. I'm just trying to find out if it's just me doing something wrong or if something may have changed in a library or something. These devices and tests seemed to work a month or so ago.

MikeyJohnston, 2025-04-14:
Again, using sscom I can send simple commands like ATI to get modem information as well as other successful location data communication via USB. But uart from esp32 is not working. I have done this with a few different baud rates now and they all work through USB, as the sim7000 auto-detects i guess?

MikeyJohnston, 2025-04-15:
I'm going to delete and re-install all of the associated libraries today to see if I can get back to a default working status.

lewisxhe, 2025-04-16:
There are two red LEDs next to the modem. Are they lit?

This has nothing to do with the library. Basic communication does not depend on the library.

Can you try this sketch again?
https://github.com/Xinyuan-LilyGO/LilyGO-T-SIM7000G/blob/master/examples/Arduino_TinyGSM/ATdebug/ATdebug.ino

MikeyJohnston, 2025-04-16:
That must be what has happened, I'm just going to get a few more boards then. The two older boards I had soldered a accelerometer and a power latching circuit. And the new board, must have had an ESD event. I probably need to 3D print a case for them and start using a ground strap. Thank you for helping me!!