Xinyuan-LilyGO/LilyGO-T-SIM7000G #300: Faulty module?

jesse377, 2025-05-18:
If i run the Arduino_NetworkTest.ino i get stuck with a signal:99. The simcard was working perfectly before in another lilygo sim 7070G. Could it be the unit is faulty? Lte antenna is connected and the simcard is supported by other lilygo sim7070g modules 

This is the response:
12:49:18.255 -> Signal:99
12:49:21.784 -> Signal:99
12:49:23.117 -> Signal:99
12:49:24.487 -> Signal:99
12:49:24.487 -> > It seems that there is no signal. Please check whether theLTE antenna is connected. Please make sure that the location has 2G/NB-IOT signal
12:49:24.487 -> SIM7000G does not support 4G network. Please ensure that the USIM card you

lewisxhe, 2025-05-19:
Is the APN set?

jesse377, 2025-05-19:
I only need sms function, so no apn is needed

lewisxhe, 2025-05-20:
Are you using 2G?
Then please send the following commands on your SIM7070G to check what the return results are
AT+CBAND?
AT+CNMP?
AT+CMNB?
AT+CPSI

jesse377, 2025-05-21:
I figured it out. When i changed it from "2 automatic" to "51 GSM and LTE" it worked. Shouldn't it also work on automatic mode? 

2 Automatic
13 GSM only
38 LTE only
51 GSM and LTE only

lewisxhe, 2025-05-22:
The automatic mode will be very slow, because SIM7000G will try all supported frequency bands. If you are sure to use GSM, you'd better fix it so that it will not need to search.