Xinyuan-LilyGO/LilyGO-T-SIM7000G #307: GPS antenna not receiving any data

AndrewLiii, 2025-07-10:
I've been trying to use different AT commands to retrieve GPS data from the stock antenna on the SIM7000G

Here's the code I used that used to work, but doesn't work now. It's very inconsistent, and now doesn't work. 

`#define TINY_GSM_MODEM_SIM7000
#define TINY_GSM_RX_BUFFER 1024
 
#define SerialMon Serial
#define SerialAT Serial1
 
#define TINY_GSM_USE_GPRS true
#define TINY_GSM_USE_WIFI false
 
#include <TinyGsmClient.h>
#include <PubSubClient.h>
#include <SPI.h>
#include <SD.h>
#include <Ticker.h>
 
// GPRS credentials
const char apn[] = "iot.1nce.net";
const char gprsUser[] = "";
const char gprsPass[] = "";
 
// MQTT broker
const char *broker = "broker.hivemq.com";
const char *topicGpsLocation = "GPS/location";
 
TinyGsm modem(SerialAT);
TinyGsmClient client(modem);
PubSubClient mqtt(client);
 
// Pins and config
#define UART_BAUD 9600
#define PIN_DTR   25
#define PIN_TX    27
#define PIN_RX    26
#define PWR_PIN   4
#define LED_PIN   12
 
uint32_t lastReconnectAttempt = 0;
 
// === GNSS Power Control ===
void enableGPS(void) {
  // Set Modem GPS Power Control Pin to HIGH ,turn on GPS power
  modem.sendAT("+CGPIO=0,48,1,1");
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("Set GPS Power HIGH Failed");
  }
  modem.enableGPS();
}
 
void disableGPS(void) {
  // Set Modem GPS Power Control Pin to LOW ,turn off GPS power
  modem.sendAT("+CGPIO=0,48,1,0");
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("Set GPS Power LOW Failed");
  }
  modem.disableGPS();
}
 
// === MQTT callback ===
void mqttCallback(char *topic, byte *payload, unsigned int len) {
  SerialMon.print("Message arrived [");
  SerialMon.print(topic);
  SerialMon.print("]: ");
  SerialMon.write(payload, len);
  SerialMon.println();
}
 
// === MQTT connect ===
boolean mqttConnect() {
  SerialMon.print("Connecting to MQTT broker: ");
  SerialMon.print(broker);
  boolean status = mqtt.connect("GPS");
  if (!status) {
    SerialMon.println(" fail");
    return false;
  }
  SerialMon.println(" success");
  return true;
}
 
void modemPowerOn() {
  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(1000);  // Datasheet Ton min = 1S
  digitalWrite(PWR_PIN, LOW);
}
 
void modemPowerOff() {
  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(1500);  // Datasheet Ton min = 1.2S
  digitalWrite(PWR_PIN, LOW);
}
 
void modemRestart() {
  modemPowerOff();
  delay(1000);
  modemPowerOn();
}
 
void setup() {
  SerialMon.begin(115200);
  delay(10);
 
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);
 
  modemPowerOn();
 
  SerialMon.println("Starting modem...");
  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);
 
  SerialMon.println("Connecting to network...");
  if (!modem.waitForNetwork()) {
    SerialMon.println("Network failed");
    while (true);
  }
  SerialMon.println("Network OK");
 
  SerialMon.print("Connecting to GPRS...");
  if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
    SerialMon.println("GPRS failed");
    while (true);
  }
  SerialMon.println("GPRS OK");
 
  mqtt.setServer(broker, 1883);
  mqtt.setCallback(mqttCallback);
}
 
void loop() {
  // Ensure network
  if (!modem.testAT()) {
    SerialMon.println("Failed to communicate with modem, restarting...");
    modemRestart();
    return;
  }
  if (!modem.isNetworkConnected()) {
    SerialMon.println("Network disconnected");
    if (!modem.waitForNetwork()) {
      SerialMon.println("Still no network");
      delay(10000);
      return;
    }
    SerialMon.println("Network reconnected");
  }
 
  if (!modem.isGprsConnected()) {
    SerialMon.println("GPRS disconnected");
    if (!modem.gprsConnect(apn, gprsUser, gprsPass)) {
      SerialMon.println("Still no GPRS");
      delay(10000);
      return;
    }
    SerialMon.println("GPRS reconnected");
  }
 
  // Ensure MQTT connection
  if (!mqtt.connected()) {
    SerialMon.println("=== MQTT NOT CONNECTED ===");
    uint32_t t = millis();
    if (t - lastReconnectAttempt > 10000L) {
      lastReconnectAttempt = t;
      if (mqttConnect()) {
        lastReconnectAttempt = 0;
      }
    }
    delay(100);
    return;
  }
 
  mqtt.loop();
 
  // === Get GNSS data ===
  SerialMon.println("Start positioning. Make sure to locate outdoors.");
  SerialMon.println("The blue indicator light flashes to indicate positioning.");
 
  enableGPS();
 
  float lat, lon;
  while (1) {
    if (modem.getGPS(&lat, &lon)) {
      SerialMon.println("The location has been locked, the latitude and longitude are:");
      SerialMon.print("latitude: "); SerialMon.println(lat, 6);
      SerialMon.print("longitude: "); SerialMon.println(lon, 6);
 
      // -----> Publish to MQTT
      char payload[64];
      snprintf(payload, sizeof(payload), "{\"lat\":%.6f,\"lon\":%.6f}", lat, lon);
      mqtt.publish(topicGpsLocation, payload);
      SerialMon.print("Published to MQTT: ");
      SerialMon.println(payload);
 
      break;
    }
    digitalWrite(LED_PIN, !digitalRead(LED_PIN));
    delay(2000);
  }
 
  disableGPS();
 
  delay(5000);  // Wait before next reading
}
 `

lewisxhe, 2025-07-10:
1. Check whether there is a 3.3V voltage output at the center point of the GPS antenna.
2. Check whether the AT command responds normally.

lewisxhe, 2025-07-11:
Thank you. Please write ATDebug, and wait for the modem to start. After sending the command AT+CGPIO=0,48,1,1, measure the voltage of these two points again. If the test result is still 0V, please send this command several times.

[image]