Xinyuan-LilyGO/LilyGO-T-SIM7000G #308: Problem getting GPS data with new antennas: example sketch and modified code hang with no NMEA output

saulfeza, 2025-07-13:
Hello,

I am having trouble getting GPS data from my LilyGO T-SIM7000G module and would appreciate any help or suggestions.

I have tested both the official Arduino example from this repository and this modified code that explicitly prints to the serial monitor whether the modem responds to restart commands:

cpp
Copiar
Editar
#define TINY_GSM_MODEM_SIM7000
#define TINY_GSM_RX_BUFFER 1024

#define SerialMon Serial
#define SerialAT  Serial1

#define PWR_PIN 4
#define PIN_TX 27
#define PIN_RX 26
#define LED_PIN 12

#define UART_BAUD 9600

#include <TinyGsmClient.h>

TinyGsm modem(SerialAT);

void modemPowerOn() {
  pinMode(PWR_PIN, OUTPUT);
  digitalWrite(PWR_PIN, HIGH);
  delay(1000);
  digitalWrite(PWR_PIN, LOW);
}

void setup() {
  pinMode(LED_PIN, OUTPUT);
  digitalWrite(LED_PIN, HIGH);

  SerialMon.begin(115200);
  delay(3000);
  SerialMon.println("=== LilyGO SIM7000 NMEA DEMO ===");

  modemPowerOn();
  delay(3000);

  SerialAT.begin(UART_BAUD, SERIAL_8N1, PIN_RX, PIN_TX);
  delay(3000);

  SerialMon.println("[INFO] Initializing modem...");
  bool restartResult = modem.restart();
  if (restartResult) {
    SerialMon.println("[RESULT] Modem restart = true");
  } else {
    SerialMon.println("[RESULT] Modem restart = false");
    SerialMon.println("[ERROR] Cannot continue. Halting...");
    while (true);
  }

  delay(3000);

  SerialMon.println("[INFO] Powering GPS on...");
  modem.sendAT("+CGNSPWR=1");
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("[ERROR] GPS Power ON command failed!");
    while (true);
  }

  SerialMon.println("[INFO] Enabling NMEA output...");
  modem.sendAT("+CGPSNMEA=1");
  if (modem.waitResponse(10000L) != 1) {
    SerialMon.println("[ERROR] NMEA output command failed!");
    while (true);
  }

  SerialMon.println("[INFO] Setup complete. Reading NMEA sentences:");
}

void loop() {
  while (SerialAT.available()) {
    SerialMon.write(SerialAT.read());
  }

  while (SerialMon.available()) {
    SerialAT.write(SerialMon.read());
  }
}
When running this code, the serial monitor shows:

csharp
Copiar
Editar
=== LilyGO SIM7000 NMEA DEMO ===
[INFO] Initializing modem...
[RESULT] Modem restart = true
[INFO] Powering GPS on...
[INFO] Enabling NMEA output...
[INFO] Setup complete. Reading NMEA sentences:
but then nothing else happens. I never receive any NMEA sentences or GPS data.

I have tried:

Using the official Arduino example from this repository (same result: it hangs waiting for GPS data).

Verifying the serial monitor speed at 115200.

Using different USB cables.

Testing multiple LilyGO T-SIM7000G boards (all behave the same).

Trying new GPS antennas (not the original ones that shipped with the module).

Testing outdoors with clear sky.

I also verified that the modem responds correctly to AT commands and confirms that it is working properly.

I am wondering if it's possible that I damaged the new antennas (for example, due to static electricity), or if they are simply incompatible or low-quality replacements.

I don't have any way to test whether the antennas actually work beyond seeing that no GPS fix or NMEA data is ever produced.

Does anyone know:

Can GPS patch antennas be damaged by static discharge?

Is there a way to test these antennas more directly?

Could this be a firmware or module configuration issue disabling NMEA output?

Is there a way to fully reset the module to factory defaults?

Any help or suggestions would be greatly appreciated.

Thank you.

lewisxhe, 2025-07-15:
```bash
AT+CGNSPWR=0
AT+CGNSPORT=3
AT+CGNSMOD=1,1,1,1
AT+CGNSCFG=2
AT+CGNSTST=1
AT+CGNSPWR=1
```

lewisxhe, 2025-07-16:
Please follow my command below to test
```bash

AT+CGPIO=0,48,1,1

OK
AT+CGNSPWR=1

OK

AT+CGNSMOD=1,1,1,0

OK
AT+CGNSPORT=3

OK
AT+CGNSMOD=1,1,1,1

OK
AT+CGNSCFG=2

OK
AT+CGNSTST=1

OK
AT+CGNSPWR=1

OK
AT+CGNSPWR?

+CGNSPWR: 1

OK
$GNVTG,,T,,M,,N,,K,N*32
$GPGSA,A,1,,,,,,,,,,,,,,,,*32
$GLGSA,A,1,,,,,,,,,,,,,,,*02
$GAGSA,A,1,,,,,,,,,,,,,,,,*23
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$GNGGA,,,,,,0,,,,,,,,*78
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,T,,M,,N,,K,N*32
$GPGSA,A,1,,,,,,,,,,,,,,,,*32
$GLGSA,A,1,,,,,,,,,,,,,,,*02
$GAGSA,A,1,,,,,,,,,,,,,,,,*23
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
$GNGGA,,,,,,0,,,,,,,,*78
$GNRMC,,V,,,,,,,,,,N*4D
$GNVTG,,T,,M,,N,,K,N*32
$GPGSA,A,1,,,,,,,,,,,,,,,,*32
$GLGSA,A,1,,,,,,,,,,,,,,,*02
$GAGSA,A,1,,,,,,,,,,,,,,,,*23
$BDGSA,A,1,,,,,,,,,,,,,,,*0F
```